lorf/csr-spi-ftdi #2: Please, help to flash CSR BC212015 via SPI (FTDI)

DJ-URAN, 2015-03-18:
Hi!
Have trouble with bluetooth module D-Link DBT-120 on CSR BC212015 chip. After some manipulations it not work in HCI mode.
I have FTDI board (FT232R chip with full pins availible). On BT module i see 4 pins SPI port: MOSI, MISO, CS, CLK. I solder them to FTDI board, install Bluesuite 2.5.8, change spilpt.dll to correct api 1.4 version for win, install libusbK drivers for FTDI, using zadig tool, connect FTDI with solder BT module to USB, run Pstool, but it could not communicate with chip. I tried  launch dfuwizard and blueflash and blueflash in cmd (with "-lpt 1" command it says error with libusb blabla...), but can't flash my chip or erase it.
Please, help! What i should do?
Thank You!
PS: I have 2 notebooks: win 7 (x86) and win 7 (x64)

lorf, 2015-03-19:
Hi! Win32 DLLs in 0.4 were incorrectly built. Sorry for that. Please try 0.4.1 zip: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.4.1

DJ-URAN, 2015-03-19:
Thank you! I just resolve my problem. see other issue "can't stop processor"

closed 2015-03-19 by DJ-URAN

P1on3R91, 2020-10-27:
> Thank you! I just resolve my problem. see other issue "error: stop processor"
Do you resolve problem with "error: stop processor" on BC212015?