lorf/csr-spi-ftdi #3: ADK3.5 BlueFlash.exe Segmentation fault and can't found device.

LuoZhongYao, 2015-04-15:
```bash
$ uname -a
Linux localhost.localdomain 3.19.3-200.fc21.x86_64 #1 SMP Thu Mar 26 21:39:42 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
```gdb
Wine-gdb> c
Continuing.
12:51:01.883: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33e67c)
fixme:msvcrt:__clean_type_info_names_internal (0x3c04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
12:51:02.063: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33dc2c)
fixme:msvcrt:__clean_type_info_names_internal (0x3d04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
12:51:03.661: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33df6c)
fixme:msvcrt:__clean_type_info_names_internal (0x3d04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
12:51:05.364: err:spi.c:120:spi_err: No FTDI device found.

Program received signal SIGSEGV, Segmentation fault.
0x003443d0 in ?? ()
Wine-gdb> bt
#0  0x003443d0 in ?? ()
#1  0x7df2f599 in spi_open () from /lib/wine/spilpt.dll.so
#2  0x7df2c53d in spifns_open_port () from /lib/wine/spilpt.dll.so
#3  0x7df2c8b5 in spifns_sequence_setvar_spiport ()
   from /lib/wine/spilpt.dll.so
#4  0x7df2cf22 in spifns_sequence_setvar () from /lib/wine/spilpt.dll.so
#5  0x7df2dcd4 in spifns_stream_sequence () from /lib/wine/spilpt.dll.so
#6  0x0034dbfb in ?? ()
#7  0x01f268ff in ?? ()
Backtrace stopped: Cannot access memory at address 0x6aec8b59
```
So I make the following change.
```diff
diff --git a/Makefile.wine b/Makefile.wine
index 22e08b7..17bc634 100644
--- a/Makefile.wine
+++ b/Makefile.wine
@@ -1,7 +1,7 @@
 TARGET_PLATFORM ?= linux
 
-WINE_LIB ?=	/usr/lib/i386-linux-gnu/wine
-I386_LIB ?=	/usr/lib/i386-linux-gnu
+WINE_LIB ?=	/usr/lib/wine
+I386_LIB ?=	/usr/lib/
 STDCXX_LIB ?=	/usr/lib/gcc/i686-linux-gnu/$(WINEGCC_VER)
 
 WINEGCC_VER := $(shell env LC_ALL=C winegcc -v 2>&1 | awk '/^gcc version/ {print $$3}' | cut -d. -f1,2)
@@ -16,9 +16,9 @@ LDFLAGS +=	-shared -m32 -static-libgcc -s
 # 64 bit winegcc tries to link with 64 bit wine libs even if -m32 is specified.
 # Direct it to the 32bit libraries.
 LDFLAGS +=	-L$(WINE_LIB) -L$(I386_LIB) -L$(STDCXX_LIB)
-#LDLIBS=	-lftdi -lusb
+LDLIBS=	-lftdi1 -lusb
 # Build with static versions of libraries
-LDLIBS +=	$(I386_LIB)/libftdi.a $(I386_LIB)/libusb.a
+#LDLIBS +=	$(I386_LIB)/libftdi.a $(I386_LIB)/libusb.a
 
 MAKEFILE =	$(firstword $(MAKEFILE_LIST))
 
diff --git a/spi.c b/spi.c
index 1308615..225b71f 100644
--- a/spi.c
+++ b/spi.c
@@ -1,4 +1,4 @@
-#include <ftdi.h>
+#include <libftdi1/ftdi.h>
 #include <stdint.h>
 #include <stdarg.h>
 #include <stdio.h>
@@ -10,6 +10,7 @@
 #endif
 #include <sys/time.h>
 #ifdef __WINE__
+#include <unistd.h> // usleep
 #include "wine/debug.h"
 #endif
 
@@ -117,8 +118,8 @@ static void spi_err(const char *fmt, ...) {
     va_start(args, fmt);
     vsnprintf(buf, sizeof(buf), fmt, args);
     LOG(ERR, buf);
-	if (spi_err_cb)
-        spi_err_cb(buf);
+	//if (spi_err_cb)
+    //    spi_err_cb(buf);
     va_end(args);
 }
```

No Segmentation fault. but can't find my device.
The shell output
```bash
fixme:winediag:start_process Wine Staging is a testing version containing experimental patches.
fixme:winediag:start_process Please report bugs at http://bugs.wine-staging.com (instead of winehq.org).
fixme:service:scmdatabase_autostart_services Auto-start service L"Parspi" failed to start: 3
12:59:01.009: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33e67c)
fixme:msvcrt:__clean_type_info_names_internal (0x3c04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
fixme:msvcp:_Locinfo__Locinfo_ctor_cat_cstr (0x33dbe8 1 C) semi-stub
fixme:msvcp:_Locinfo__Locinfo_ctor_cat_cstr (0x33da18 1 C) semi-stub
12:59:01.226: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33dc2c)
fixme:msvcrt:__clean_type_info_names_internal (0x5404e0) stub
12:59:03.903: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33ae4c)
fixme:msvcrt:__clean_type_info_names_internal (0x5404e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
12:59:05.008: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33df6c)
fixme:msvcrt:__clean_type_info_names_internal (0x3d04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
12:59:06.257: err:spi.c:120:spi_err: No FTDI device found
12:59:08.539: warn:basics.cpp:242:spifns_enumerate_ports: No FTDI device found, calling port enum callback (1, "No FTDI device found", 0x33ae4c)
fixme:msvcrt:__clean_type_info_names_internal (0x3d04e0) stub
fixme:msvcrt:__clean_type_info_names_internal (0x3d9450) stub
```

lorf, 2015-04-15:
Hi! Do You have permissions for FTDI device? It should be set up in udev config, like this:

    cat <<_EOT_ | sudo tee -a /etc/udev/rules.d/99-ftdi.rules
    # FT232RL
    SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", MODE="0660", GROUP="plugdev"
    _EOT_
    sudo usermod -a -G plugdev `id -un`

You will need to relogin or run commands under "newgrp plugdev" for user change to take effect.

lorf, 2015-04-16:
Can You also post output of lsusb and output of the following command?

    env FTDI_LOG_LEVEL=debug wine /path/to/BlueFlashCmd.exe -chipver

LuoZhongYao, 2015-04-17:
i' m sorry.I uninstalled this environment, the use of virtual machines to
run blueflsah.

LuoZhongYao, 2015-04-17:
This is one of my mistakes, I am using a USB <=> SPI device instead FTDI
device.

2015-04-17 8:46 GMT+08:00 一二 <luozhongyao@gmail.com>:

lorf, 2015-04-17:
OK. I'll investigate the segfault, thanks for reporting!

Makcumus, 2015-08-12:
Sorry for offtop. 
FTDI in new driver reset PID of non original (chinese) chips 232R to 0000, and i think it will be usefull add to lib this PID, or notes in manual. Of course it easy to return PID back by using FTDI`s MProg. Have a nice day.

lorf, 2015-08-12:
Thanks, @Makcumus. I'll look into it.

Makcumus, 2015-08-12:
Good day.

Also using Mprog 3.5 for FTDI232R you must set options: 
1) bus powered current 500ma 
2) I/o high current
Else you get error -> cant stop processor 
And Dll v1.4 is not working with BlueSuite 2.6.0 on WIN7 x64 - programm error.
It looks like, that there is no enough current from some USB (for some PC it is 500ma only in some it more than 1Amp), may be it will be usefull to use external power (and\or self powered option in Mprog). In my case i use double USB cable.

>
>Thanks, @Makcumus . I'll look into it.

С уважением,
m m
makcum1982@mail.ru

DJ-URAN, 2015-08-12:
Confirm. 1.4v dll not work in bluesuite 2.6.0 (and 2.5.8 also).
I use this dll with BlueLab 4.1 with no connection problems.

lorf, 2015-08-13:
@Makcumus @DJ-URAN Can you please show the debug output of the command producing error? Assuming it is run on Windows:
    set FTDI_LOG_LEVEL=debug
    blueflashcmd -dump test (or anything producing an error)

lorf, 2015-08-15:
I tried but cannot reproduce error in BlueSuite 2.5.8 and 2.6.0 with api1.4 dll from csr-spi-ftdi-0.4.2, counterfeit FT232RL chip and HC05 module. Tried in Win7/x32 on a VM and in Win8/x64 on real HW.

@Makcumus @DJ-URAN - can You help me with this? Please try to reproduce the error with command like this:

blueflashcmd -trans "SPIDEBUG=ON SPIDEBUG_FILE=c:\temp\debug.log" -dump test

and send resulting log to dmitry.frolov@gmail.com

lorf, 2015-08-15:
Other things to try:
* Lower FTDI_BASE_CLOCK, e.g. set in environment:

    set FTDI_BASE_CLOCK=1000000
* Lower resistor values in SPI lines

lorf, 2015-08-15:
@Makcumus WRT USB bus current, AFAIK most motherboards just ignore the MaxPower setting of USB device and supply as much current as they can. Simple setup with FT232RL and HC05 module should not draw more than 150mA, so it shouldn't be an issue.

Issue may arise if You power HC05 from FT232R internal 3.3V regulator. Current consumption will rise on flash write operations, and it may exceed FT232R 50mA limit.

DJ-URAN, 2015-08-15:
lorf, to help you, should i install CSR drivers first (for DFU) and then try your command in cmd? Should i switch first my FTDI driver to WinUSB in Zadig?

lorf, 2015-08-16:
No, please, don't install CSR USB drivers, and select LPT (not USB) transport at installation time, otherwise You'll need to specify "-LPT 1" option on command line.

You may try with libusbK driver first and if error occurs, try to switch to WinUSB.

DJ-URAN, 2015-08-16:
lorf! I tried make dump in blueflashcmd, but it give me ERROR: <filename> not given for "-dump
maybe mistake in command?

DJ-URAN, 2015-08-16:
PS: when i install bluesuite on my pc it said that LPT not supported for 64 bit systems (i have x64 windows)/

lorf, 2015-08-17:
You need to supply a filename to the -dump option, like this:
    blueflashcmd -dump somefilename
At the end You should get two files, `somefilename.xpv` and `somefilename.xdv`.

It's OK when BlueSuite complains about LPT on x64 systems, since we actually will not use LPT port nor LPT driver, but provide our own instead.

DJ-URAN, 2015-08-17:
I try your command blueflashcmd -trans "SPIDEBUG=ON SPIDEBUG_FILE=c:\temp\debug.log" -dump test. It contains -dump TEST at the end. But nothing changed.

lorf, 2015-08-17:
It looks like filename "TEST" is magic and doesn't work!

lorf, 2015-08-28:
After debugging @DJ-URAN issue over email, we found the following:

1. Support for BlueCore 2 chips (BC212015 in particular) was dropped in BlueSuite 2.4. So the newest BlueSuite supporting BC2 is BlueSuite 2.3 (use spilpt.dll from `csr-spi-ftdi-*/spilpt-*-api1.3` with it).
2. Communicating with BC2 over SPI requires lower SPI speeds. You may try:

        blueflashcmd -trans "SPITRANS=LPT SPIPORT=1 FTDI_BASE_CLOCK=400000" -identify

   and play with FTDI_BASE_CLOCK. Default for FTDI_BASE_CLOCK in csr-spi-ftdi-0.4.2 is 4000000. For @DJ-URAN the following command from BlueSuite 2.3 with `csr-spi-ftdi-0.4.2/spilpt-win32-api1.3/spilpt.dll` worked:

        blueflashcmd -trans "SPITRANS=LPT SPIPORT=1 FTDI_BASE_CLOCK=400000" -dump test1

lorf, 2016-07-14:
Looks like all problems were addressed so I'm closing it

closed 2016-07-14 by lorf