lorf/csr-spi-ftdi #4: Unable to start read (invalid control data)

milkpirate, 2015-08-28:
Hey,

I'm trying to dump (and flash) my HC-05 with the HID firmware. I followed the "Installing on Windows" and used the prebuild dll (v1.4 + BlueLab41) and it spits this out:

```
D:\Downloads\CSR\BlueLab41>BlueFlashCmd.exe -TRANS "SPITRANS=LPT SPIPORT=1" -DUMP OriginalFirmware
CSR BlueFlash Tool (Command Line) BlueSuite-2.1.1-Release

Copyright (C) 2002-2008 Cambridge Silicon Radio Ltd.

11:41:04.462: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.480: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.496: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.511: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.527: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.559: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.592: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.608: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.624: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.658: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.704: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
11:41:04.739: err:basics.cpp:554:spifns_sequence_read: Unable to start read (invalid control data)
...
```
My wiring corresponds to "Using FT232RL breakout board as a programmer" and looks like:
[image]

Could you give me a starting point to check whats wrong? Wiring? FTDI? HC-05? Software? Driver?

Thank you very much!

lorf, 2015-08-28:
Hi! Is this the same adapter as SparkFun [FT232R Breakout](https://www.sparkfun.com/products/retired/718)? From what I can realize You connected SCK to VCCIO pin of the adapter (pin 4, counting from TX LED), while you need to connect it to RTS pin (pin 3 from TX LED). Or am I seeing it wrong?

milkpirate, 2015-08-29:
Hey lorf,

yes it is. I changed the size of the SMD parts from 0603 to 0805 (but didnt think of not getting the PTC in the right rating and size around here). Its really likely that I made the wiring wrong - tinkering evening was really frustrating and I remembered that I have a Stellaris Lauchpad and went over to that sollution: [Frans-Willem](https://github.com/Frans-Willem/CsrUsbSpiDeviceRE) - which worked like a charm (no hazzel with replacing dll's and stuff).

But youre really doing great work here! I'm so greatful that someone is doing that hacking/coding for people like me. Thank you very much!

lorf, 2015-08-29:
Glad You have a working setup! No, sorry, I don't have those images.

closed 2015-08-29 by lorf