lorf/csr-spi-ftdi #7: FT2232HL not init

jldiez, 2016-06-18:
Hello, i'm trying connect to csr8645 via ft2232hl board (0x0403:0x6010). I followed your indications, install libusbk drivers, etc, also i activated ftdi debug, but i don't know what it's happening. Could you help me? I paste the debug stdout on windows 7, i also tested in windows xp, but the problem it's the same.

OUTPUT:

C:\Program Files (x86)\CSR\BlueSuite 2.6.2>blueflashcmd -trans "SPITRANS=USB SPI
PORT=1 SPIDEBUG=ON SPIDEBUG_FILE=C:\csr-debug.log FTDI_LOG_LEVEL=debug,dump FTDI
_LOG_FILE=C:\csr-spi-ftdi-debug.log" -identify
blueflashcmd, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2002 - 2015.

13:21:50.962166: info:basics.cpp:103:spifns_init: Detected SPI API version 0x010
4, using version 0x0104
13:21:50.963167: debug:spi.c:525:spi_init: spi_nrefs=0, spi_dev_open=0
13:21:50.963167: all:spi.c:534:spi_init: csr-spi-ftdi 0.5.0, git rev 92e495e

13:21:50.981168: debug:spi.c:487:spi_enumerate_ports: find all: 0x0403:0x6001
13:21:50.998169: debug:spi.c:487:spi_enumerate_ports: find all: 0x0403:0x0000
13:21:51.014169: debug:spi.c:487:spi_enumerate_ports: find all: 0x0403:0x6010
13:21:51.032170: err:spi.c:505:spi_enumerate_ports: FTDI: ftdi_usb_get_strings()
 failed: libusb_get_string_descriptor_ascii() failed
13:21:51.033171: debug:spi.c:567:spi_deinit: spi_nrefs=1, spi_dev_open=0
13:21:51.034171: debug:basics.cpp:137:spifns_get_last_error
13:21:51.034171: debug:basics.cpp:147:spifns_clear_last_error
13:21:51.034171: debug:spi.c:567:spi_deinit: spi_nrefs=0, spi_dev_open=0
==============================================================================
ERROR: FTDI: ftdi_usb_get_strings() failed: libusb_get_string_descriptor_ascii()
 failed
==============================================================================
Failed

Thank you very much for you help.

jldiez, 2016-06-20:
Hello,

I've tested in Ubuntu with wine following yours indications and all worked ok, only one issue, i had to connect the GND of FT2232H with GND of CSR8645 because showed the message "Unable to start read (invalid control data)" and the read was very slow or impossible. After to connect two GND's all Ok (I use two independent power supplies). The question is, Why in a real Windows not runs?

Thanks.

lorf, 2016-06-21:
Hi! Can You, please, try the [0.5.1-a1 version](https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.1-a1) I uploaded today in Windows and see if it fixes the issue?

I don't have any FT2232 chips to try so can only guess what happens. There is [libusb bug report]( http://marc.info/?l=libusb-devel&m=144364532621752&w=2) with similar symptoms in Windows caused by unusual usb device configuration. Also FTDI chip EEPROM config can disable reporting of the serial number to the host (see `SerNumEnable7` in [D2XX Programmers  Guide](http://www.ftdichip.com/Support/Documents/ProgramGuides/D2XX_Programmer's_Guide(FT_000071).pdf) which can cause ftdi_usb_get_strings() function to fail. In 0.5.1-a1 I implemented a workaround for the second, please try and report.

lorf, 2016-06-21:
BTW, I added GND connection to the table in the README, thanks!

jldiez, 2016-06-21:
Hi, it works!

I tested in Windows XP. This is the output of blueflashcmd:

C:\Archivos de programa\CSR\BlueSuite 2.6.2>BlueFlashCmd.exe chipver
blueflashcmd, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2002 - 2015.

18:40:18.149726: all:spi.c:550:spi_init: csr-spi-ftdi 0.5.1-a1, git rev fdccfee
18:40:18.962226: warn:spi.c:496:spi_enumerate_ports: FTDI: ftdi_usb_get_strings(
) failed: [-9] libusb_get_string_descriptor_ascii() failed
GBL_CHIP_VERSION = 0x03320000
Result:
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.1-a1 (git rev fdccfee)
Time open: 0.12 s
Time in xfer: 0.01 s (12.00% of open time)
Reads: 8 (16 bytes, 2.00 bytes avg read size)
Writes: 4 (12 bytes, 3.00 bytes avg write size)
Xfer data rate: 1.82 KB/s (28 bytes in 0.01 s)
IOPS: 800.00 IO/s (12 IOs in 0.01 s)
FTDI chip: FT2232H (4), buffer size: 8192 bytes
FTDI stats: 1333.00 xfers/s (0.00 short reads/s,
            20 xfers/0 short reads in 0.01 s,
            1.00 xfers/IO, 48.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Success

C:\Archivos de programa\CSR\BlueSuite 2.6.2>
-------------------------------------------------------------

Although the failure appears "failed: [-9]" works, I also read the chip with PsTool and all was ok.

Thank you very much.

lorf, 2016-06-22:
Yes, I turned the error into a warning, since it's not critical to operation. Thanks for testing!

closed 2016-06-25 by lorf

crashracer, 2018-06-17:
Just make a cable to connect to USB_DN, USB_DP, GND, with 3v. Forget the SPI interface.

commit fe6247ac (2016-06-25): Turn get FTDI strings error into a warning.

Reporting of the chip serial number to the host can be disabled in FTDI
EEPROM config (see SerNumEnable* variables in FT2XX Programmer's Guide).
This can cause ftdi_usb_get_strings() to fail (specifically on Windows).
Turn this failure into a warning, since it's not critical. This fixes #7.

Also log libftdi error codes.