lorf/csr-spi-ftdi #8: Resetting XAP

nodep, 2016-06-21:
Hello,

I am trying to reconfigure a CSR8635. I have a cheap (probably counterfeit) FT232 module and I create 1.8v for the VCCIO with a voltage divider (I don't have a 1.8V regulator).

I've connected everything, started the programmer and this is what I get for identify:

----------------------------------
C:\Users\nodep>blueflashcmd -identify
blueflashcmd, version 2.3.0.15
Copyright (C) 2002-2010, Cambridge Silicon Radio Ltd.
17:47:53.273164: all:spi.c:534:spi_init: csr-spi-ftdi 0.5.0, git rev 92e495e

Resetting XAP
----------------------------------

And it just hangs there. I waited more then 10 minutes. Sometimes I get several lines of:

----------------------------------
17:48:08.526063: err:basics.cpp:446:spifns_sequence_read: Unable to start read (invalid control data)
17:48:08.531068: err:basics.cpp:446:spifns_sequence_read: Unable to start read (invalid control data)
17:48:08.536697: err:basics.cpp:446:spifns_sequence_read: Unable to start read (invalid control data)
17:48:08.542121: err:basics.cpp:446:spifns_sequence_read: Unable to start read (invalid control data)
----------------------------------

The funny thing is that the SPI is very busy all this time. I've checked both with a scope and logic analyzer.

I could send you a Saleae capture of the traffic on the SPI, if that helps.

Thank you!

lorf, 2016-06-22:
You really need a stable power supply for VCCIO. You can get 1.8V from CSR8635 module for example.

1V8 CSR chips accept up to 3.0V on SPI pins (e.g. official CSR programmer uses 3.0V), so You can use 2 generic 1.5V batteries for power supply, but be sure to connect SPI pins to CSR chip via resistors (1K or lower).

Other options to try, but not recommended: try connecting electrolytic capacitor, 100uF or up, from VCCIO to GND, lower divider resistor values.

nodep, 2016-06-22:
Thanks a lot for help and this project!

I think I will just build a proper dedicated programmer. I took too many shortcuts and a didn't have enough patience.

I will make a PCB based on your schematic.

Thanks again!

closed 2016-06-22 by nodep