lorf/csr-spi-ftdi #9: Addressing more that one of these on a single computer.

T-2, 2016-07-07:
I put together one of these and it works great under Linux/Wine. Thank you very much. I have an application that will require two of these tied to one computer. Is there a way to address each individually with the various CSR tools?

lorf, 2016-07-12:
Currently no, sorry. Stream handling have to be implemented in csr-spi-ftdi for this to work and it also may require some USB device handling tweaks. BlueSuite supports multiple programmers since 2.4 afaik. For other tools - I don't know.

T-2, 2016-07-12:
Thanks for getting back to me on this.

I went ahead and built a second one just to see what happens. Tested with pscli, blueflashcmd, and PSTools all running under wine. Both interfaces work by themselves if only one is plugged into the computer at a time as you would suspect. Things get a tad weird if you have both plugged in at the same time. With all 3 CSR (version 2.6.2) tools I can talk to one of the interfaces but it errors out if I try to talk to the second one. PSTools shows both interfaces by their serial number so you can pick the one you want at startup but only one actually works.

With pscli you can differentiate the two interfaces by pscli.exe -TRANS "SPITRANS=USB SPIPORT=1" <command> or SPIPORT=2. However only one will work.

Similarly for BlueFlashCmd use  BlueFlashCmd.exe -usb 1 <command> or -usb 2. Only one works there also.

It appears that all cases the first interface you plug in is the the one that will work.

It looks like it my be possible to switch between the interfaces one at a time if I can find a way from a script to disconnect the individual USB port at a one at a time basis.

Anyway if you decide you want to pursue getting this possible feature to work I will help in anyway I can. Just let me know.

T-2, 2016-07-27:
I was reviewing your answer to to my query a few weeks ago. I don't know what you mean by "Stream handling". If you mean being able to talk to more than one of these interfaces at the same time that is not required by my application. I just need 2 interfaces plugged into the same computer at the same time however I only need to use one at a time. So I just need a way in a script to switch to the interface I need at the time, do stuff with that one then disable the first one and enable the second one and then do other stuff with the second one. Could that be done more easily in your driver then "stream handling?

lorf, 2016-07-29:
Hi! By "stream handling" I meant a functionality that BlueSuite expects from the driver. Thought It needed some work to implement.

In your situation as a workaround You may have luck with disabling/enabling USB devices as needed in your scripts, since csr-spi-ftdi selects the first FTDI device it sees. I didn't tried that, however.