lorf/csr-spi-ftdi #10: Can read chipver, but identity and other blueflashcmd commands fail

juancgarcia, 2016-08-08:
I'm using an FT232RL based FTDI adapter called the [USB BUB](http://web.archive.org/web/20150921035219/https://moderndevice.com/product/usb-bub-i/) to communicate with a HC-05 module. I'm pretty sure I have it wired up properly as I am getting some promising debug output from chipver, but it's inconsistent.

blueflashcmd ... chipver

debug output:
> [**********][Timestamp since epoch]
> [1808698280][TID: 247208][SID: 01] 
> [1808698280]read ff9a 4826 
> [1808698286]read ff9a 4826 
> [1808698287]Core type bluecore
> [1808698287]FF9A=0x4826 FE81=0x00FF
> [1808698287]Chip type = 4, Family = 1, DisplayName="BC4-EXT (cyt)"
> [1808698287]Pre BC7
> [1808698287]bluecore_chip

console out:

> blueflashcmd, version 2.6.0.450 Release
> Copyright Cambridge Silicon Radio Limited 2002 - 2015.
> 
> 16:29:02.243486: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.1, git rev b60665a
> GBL_CHIP_VERSION = 0x4826
> Result:
> *** FTDI Statistics ********************************************************
> csr-spi-ftdi version: 0.5.1 (git rev b60665a)
> Time open: 0.03 s
> Time in xfer: 0.01 s (34.38% of open time)
> Reads: 4 (8 bytes, 2.00 bytes avg read size)
> Writes: 2 (6 bytes, 3.00 bytes avg write size)
> Xfer data rate: 1.24 KB/s (14 bytes in 0.01 s)
> IOPS: 545.00 IO/s (6 IOs in 0.01 s)
> FTDI chip: FT232R (3), buffer size: 384 bytes
> FTDI stats: 1181.00 xfers/s (90.00 short reads/s,
>             13 xfers/1 short reads in 0.01 s,
>             2.00 xfers/IO, 37.00 bytes/xfer)
> SPI max clock: 25 kHz, min clock: 25 kHz, slowdowns: 0
> ****************************************************************************
> Success

However it sometimes fails altogether

debug:

> [**********][Timestamp since epoch]
> [1807884228][TID: 243504][SID: 01] 
> [1807884228]read ff9a  Failed
> [1807884232]read ff9a  Failed
> [1807884236]read ff9a  Failed
> [1807884239]read ff9a  Failed
> ...
> [1807884337]read ff9a  Failed
> [1807884342]read ff9a 4c00 
> ...
> [1807884421]read ff9a  Failed
> [1807884425]read ff9a  Failed

console:

> 16:15:28.195339: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.1, git rev b60665a
> 16:15:28.378221: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:15:28.382224: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ...
> 16:15:28.570839: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:15:28.574843: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> blueflashcmd, version 2.6.0.450 Release
> Copyright Cambridge Silicon Radio Limited 2002 - 2015.
> 
> ==============================================================================
> ERROR: Error detecting chip type (Unable to start read (invalid control data))
> ==============================================================================
> Failed
> *** FTDI Statistics ********************************************************
> csr-spi-ftdi version: 0.5.1 (git rev b60665a)
> Time open: 0.21 s
> Time in xfer: 0.00 s (2.34% of open time)
> Reads: 63 (126 bytes, 2.00 bytes avg read size)
> Writes: 62 (186 bytes, 3.00 bytes avg write size)
> Xfer data rate: 60.94 KB/s (312 bytes in 0.00 s)
> IOPS: 25000.00 IO/s (125 IOs in 0.00 s)
> FTDI chip: FT232R (3), buffer size: 384 bytes
> FTDI stats: 25600.00 xfers/s (0.00 short reads/s,
>             128 xfers/0 short reads in 0.00 s,
>             1.00 xfers/IO, 83.00 bytes/xfer)
> SPI max clock: 50 kHz, min clock: 50 kHz, slowdowns: 0
> ****************************************************************************

I get this output from identify

debug:

> [**********][Timestamp since epoch]
> [1809817385][TID: 247924][SID: 01] 
> [1809817385]read ff9a 4826 
> [1809817389]read ff9a 4826 
> [1809817390]Core type bluecore
> [1809817390]FF9A=0x4826 FE81=0x00FF
> [1809817390]Chip type = 4, Family = 1, DisplayName="BC4-EXT (cyt)"
> [1809817390]Pre BC7
> [1809817390]bluecore_chip
> [1809817394]write 006a 0003 
> [1809817397]write 006a 0003 
> [1809817397]write 006a 0003 
> [1809817397]write 006a 0003 
> [1809817398]write 006a 0003 
> [1809817409]write 006a 0003 
> [1809817409]write 006a 0003 
> [1809817409]write 006a 0003 
> [1809817409]write 006a 0003 
> [1809817418]write 006a 0003 
> [1809817422]read 0073 0082 
> [1809817424]read 0073 0082 
> [1809817426]read 0073 0082 
> [1809817428]read 0073 0082 
> [1809817430]read 0073 0082 
> [1809817432]read 0073 0082 
> [1809817434]read 0073 0082 
> [1809817436]read 0073 0082 
> [1809817438]read 0073 0082 
> [1809817440]read 0073 0082 
> [1809817442]verified write a000 0000 
> [1809817444]verified write a001 0000 
> [1809817446]verified write a002 00e0 
> [1809817450]read ffe9 0000 
> [1809817450]read ffe9 0000 
> [1809817450]verified read ffe9 0000 
> [1809817452]verified write ffe9 0054 
> [1809817454]verified write ffea 0000 
> [1809817458]read ff7e 0000 
> [1809817458]read ff7e 0000 
> [1809817458]verified read ff7e 0000 
> [1809817460]verified write ff7e 0000 
> [1809817462]verified write ffde 0000 
> [1809817466]read ffe8 0000 
> [1809817466]read ffe8 0000 
> [1809817466]verified read ffe8 0000 
> [1809817469]verified write ffe8 0000 
> [1809817471]verified write 0076 0002 
> [1809817471]write ff91 0005 
> [1809817474]verified write 0077 0001 
> [1809817477]verified write 006a 0000 
> [1809817479]read ff91 ffff 
> [1809817481]read ff91 0000 
> [1809817483]read ff91 0000 
> [1809817483]write 006a 0002 
> [1809817483]write 006a 0002 
> [1809817483]write 006a 0002 
> [1809817483]write 006a 0002 
> [1809817486]write 006a 0002 
> [1809817486]write 006a 0002 
> [1809817486]write 006a 0002 
> [1809817486]write 006a 0002 
> [1809817489]write 006a 0002 
> [1809817489]write 006a 0002 
> [1809817489]write 006a 0002 
> [1809817489]write 006a 0002 
> [1809817491]write 006a 0002 
> [1809817491]write 006a 0002 
> [1809817491]write 006a 0002 
> [1809817491]write 006a 0002 
> [1809817491]write 006a 0002 
> [1809817493]write 006a 0002 
> [1809817493]write 006a 0002 
> [1809817493]write 006a 0002 
> [1809817512]write 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
> [1809817512]write 006a 0002 
> [1809817514]write 006a 0002 
> [1809817519]read ffeb 00ff 
> [1809817519]read ffeb 00ff 
> [1809817519]verified read ffeb 00ff 
> [1809817523]read ffec ffff 
> [1809817523]read ffec ffff 
> [1809817523]verified read ffec ffff 
> [1809817525]write ffe0 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 
> [1809817528]verified write ffeb 00ff 
> [1809817531]verified write ffec ffff 
> [1809817532]write f9c0 0009 
> [1809817532]write f9c0 0009 
> [1809817532]write f9c0 0000 
> [1809817532]write f9c0 0000 
> [1809817543]write f9c0 0000 
> [1809817547]write f9c0 0009 
> [1809817547]write f9c0 0009 
> [1809817547]write f9c0 0000 
> [1809817547]write f9c0 0000 
> [1809817560]write f9c0 0000 
> [1809817562]write 006a 0003 
> [1809817565]write 006a 0003 
> [1809817565]write 006a 0003 
> [1809817565]write 006a 0003 
> [1809817565]write 006a 0003 
> [1809817576]write 006a 0003 
> [1809817576]write 006a 0003 
> [1809817576]write 006a 0003 
> [1809817576]write 006a 0003 
> [1809817587]write 006a 0003 
> [1809817593]read 0073  Failed
> [1809817596]read 0073  Failed
> [1809817599]read 0073  Failed
> ...
> [1809817940]read 0073  Failed
> [1809817943]read 0073  Failed

console:

> blueflashcmd, version 2.6.0.450 Release
> Copyright Cambridge Silicon Radio Limited 2002 - 2015.
> 
> 16:47:41.452445: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.1, git rev b60665a
> Resetting XAP
> Identifying XAP
> 16:47:41.739729: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:47:41.744225: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:47:41.747226: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ...
> 16:47:42.098902: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:47:42.102904: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 16:47:42.106408: err:spi.c:118:spi_ftdi_xfer: FTDI: write data failed: [-1] usb bulk write failed
> ^C

But again, as with chipver, it sometimes completely fails without any successful reads/writes.

I'm hoping you can help shed some light on the issue.

For further reference, here are some bad photos of my wiring setup.

[image]
[image]

lorf, 2016-08-10:
These sympthoms are most likely caused by SPI signals (CLK signal in
particular) aberrations.

Please try connecting ~1 nF ceramic capacitor in parallel with resistor
going to HC-05 CLK pin. Or eliminate that resistor completely, connecting
FT232 pin directly to HC-05 CLK pin.

juancgarcia, 2016-08-13:
Eliminating the resistor had no effect. I've ordered some capacitors to see if that might solve the issue.

lorf, 2016-08-14:
Can You, please, try without resistors, connecting all pins directly? And, please, check all connections.

Also seen a USB error (FTDI: write data failed: [-1] usb bulk write failed), so please check USB connection and cable.

juancgarcia, 2016-08-14:
I had previously tried bypassing all the resistors with no luck.

I'm pretty sure the that FTDI message is related to me hitting ctrl-C for quitting the command.

This may be relevant, though. I'm testing on a Windows 10 laptop, and a Windows 7 desktop. I recently read in some articles and forums that the CSR suite doesn't work well past WinXP.

lorf, 2016-08-14:
OK, can You, please, try with lower SPI speed like this:

```
set SPIMAXCLOCK=100
blueflashcmd identify
```

juancgarcia, 2016-08-14:
100 didn't give any change, but when I pushed it down to 10, `identify` gave a new message in the console output:

> ...
> 10:50:54.274737: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ...
> Unable to start read (invalid control data)
> 10:50:55.158399: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ==============================================================================
> ERROR: Could not stop XAP
> ==============================================================================
> 10:50:55.188445: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ...
> 10:50:55.380581: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> 10:50:55.390588: err:basics.cpp:547:spifns_bluecore_xap_stopped: No reply from XAP
> Device 0: SPI error.  Removed device.
> 10:50:55.404597: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ...
> 10:50:55.584218: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
> ==============================================================================
> ERROR: Device 0: SPI error.  Removed device.
> ==============================================================================
> *** FTDI Statistics ********************************************************
> csr-spi-ftdi version: 0.5.1 (git rev b60665a)
> Time open: 3.83 s
> Time in xfer: 0.51 s (13.47% of open time)
> Reads: 345 (690 bytes, 2.00 bytes avg read size)
> Writes: 702 (2163 bytes, 3.08 bytes avg write size)
> Xfer data rate: 5.39 KB/s (2853 bytes in 0.51 s)
> IOPS: 2025.00 IO/s (1047 IOs in 0.51 s)
> FTDI chip: FT232R (3), buffer size: 384 bytes
> FTDI stats: 1773.00 xfers/s (326.00 short reads/s,
>             917 xfers/169 short reads in 0.51 s,
>             0.00 xfers/IO, 107.00 bytes/xfer)
> SPI max clock: 10 kHz, min clock: 10 kHz, slowdowns: 0
> ****************************************************************************
> Failed

lorf, 2016-08-15:
10 kHz is too low, some things are time dependant and programmer misses something with this speed.

These communication errors also may be caused by unstable power supply. How do You power the HC-05? You shouldn't power it from FT232 3.3V regulator, it can be too weak for HC-05.

If You do power it from FT232 3.3V source, You can try connecting some decoupling capacitors to power lines (like 100nF ceramic and 100uF electrolytic) as close to HC-05 as possible. But better use some other power supply (e.g. two AA batteries).

juancgarcia, 2016-08-15:
I've been using a salvaged LF33 to take 5V/1A to 3.3v. It seems to power the HC-05 well enough for normal bluetooth serial operation, but perhaps programming draws more current than it can handle.

A couple of days ago, I decided to assume everything might be faulty, and ordered replacements for every piece of this puzzle. I have incoming a proper 3.3+5V breadboard power supply, a new FT232RL breakout, and another HC-05 module. I even ordered a parallel port adapter cable (I found my desktop motherboard has an unused LPT header) to try and eliminate the ft232 as a baseline test.

I'll let you know how I fare once everything comes in.

juancgarcia, 2016-08-16:
Eureka! Believe it or not, it was my FTDI module that was causing the problem. My new (probably counterfeit) FT232RL breakout arrived today, and once I substituted it into my breadboard, everything started to work as expected. I've been able to dump/upload firmware read/write persistent storage, everything.

I can only assume that maybe my ft232rl is such an old revision of the chip that isn't suitable for this communication. Standard serial functions and programming Arduinos has never been an issue. In fact it continues to work in that capacity.

lorf, 2016-08-16:
Great! I just googled a bit and found a [known problem](http://blog.bitheap.net/2012/03/ft232r-bitbang-mode-is-broken.html) with rev A and B of FT232RL in bitbang mode (which is used in csr-spi-ftdi).

Can You please tell what revision of FT232RL You have on USB BUB? It's on the top of the chip after date code. I want to add that info to the docs.

juancgarcia, 2016-08-16:
Looks like Revision B [0822-B]

mkeyno, 2018-09-03:
@juancgarcia  did you get any success using parallel port instead of FT232?

juancgarcia, 2018-09-24:
@mkeyno I didn't pursue the lpt port once I found the replacement FT232 solved my issue.

mkeyno, 2018-09-28:
@juancgarcia  I did solve my parallel port issue after find out most of  the FT232  boards are fake