lorf/csr-spi-ftdi #11: Different pinout on Bluetooth Module and FT232 not being recognized.

Limoster, 2016-10-06:
According to your pinout ( https://github.com/lorf/csr-spi-ftdi/blob/master/README.md#using-ft232rl-breakout-board-as-a-programmer ) I don't know how to plug the pins in the CSR8635.
[image]

This is my pinout and in my TDA7492P module there are 7 pads right under the BT module. From left to right: **SPI_PCM, GND, VBAT, MOSI, CLK, CSB (?), MISO.**

**How should I connect those pads?**

Also my FT232 module is not being recognised. Downloaded MProg to look if the ID was erased to 0000 but it's still 6001 and it seems to recognise it. But PSTools can't find it. Installed drivers from zadig's program like you recommended.

lorf, 2016-10-07:
Assuming your board is already powered from some power source, connections should look like this:

**CSR8635 module** - **FT232**
1V8 - VCCIO
GND - GND
SPI_PCM - Connect to 1V8 via 10k resistor
MOSI - RI#
CLK - RTS#
CSB - DTR#
MISO - DSR#

The 3V3/5V jumper on FT232 module should be removed, or You will fry CSR chip.

As for libusb driver, please read some tutorials, e.g. [here](https://mitxela.com/projects/bluetooth_hid_gamepad), [here](http://embedded-funk.net/running-libftdi-under-windows/).

Limoster, 2016-10-07:
Should I add 220ohm resistors to any pins? And can I use the 5v/3.3v VCC
from FT232 to power SPI_PCM with 10kohm resistor?
SPI_PCM has pad type Weak Pull Down Resistor. 0=PCM/PIO and 1=SPI.

lorf, 2016-10-08:
Resistors are recommended at least at MISO pin. You can omit them ony if You are sure of connections.

Maxumum voltage at SPI_PCM is 3.6V according to a datasheet, so yes, I think You can connect it to a 3.3V via resistor.

Limoster, 2016-10-08:
By the way, there is no DSR# on FT232. Only RSD#. You might've read it
backwards.

Limoster, 2016-10-08:
I have access to FT232 now. My issue was I didn't replace usbspi.dll on installation folder. Had to do it also for CSR8600 ROM Series Configurator Tool and now it works. THANKS!!!

lorf, 2016-10-08:
DSR (Data Set Ready) is one of RS-232 standard signals. On the other side I've seen "RDS" mark on the counterfeit FT232RL module.

Limoster, 2016-10-08:
hahaha 👍 No wonder it was only 0.6€ on Aliexpress.
It works like a charm and ftdi hasn't blocked the chip.
It might be a very well made counterfeit die.

Limoster, 2016-10-13:
One final question: I was jerking around CSR 8600 ROM Series Configuration Tool  and I found you can add tones and voice prompts. I thought I was able to make a small 3s WAV or MP3 file under 100KB and put it inside the eeprom (mine is empty so I have aprox 128KB free). The programming utility almost got it in the eeprom bun when E2BLUECMD tied to flash the eeprom, it failed miserably.

Is that because the chip was in SPI instead of PCM? And if so, how can I get it running?

PD: Instead of adding a Power On tone, my idea is to make a voice promt that starts when the device powers on. It seems like the programming utility is able to do that and the chip has an MP3 decoder.

lorf, 2016-10-13:
No idea, sorry.

closed 2017-11-14 by lorf