lorf/csr-spi-ftdi #12: Invalid control data...

d0narnbl1tz, 2016-11-11:
Hi, can i get an hint maybe whats going wrong maybe

[image]
[image]
Yep the cables are servo cables but i didnt find other connectors this evening :smile:

lorf, 2016-11-14:
Hi! Do you have photos/links for a bluetooth board? Which module/chip it uses? What is the pinout of pads You soldered to? Did You [enabled SPI port](https://github.com/lorf/csr-spi-ftdi#chip-notes) on a CSR chip?

lorf, 2016-11-14:
Hi! Do you have photos/links for a bluetooth board? Which module/chip it uses? What is the pinout of pads You soldered to? Did You [enabled SPI port](https://github.com/lorf/csr-spi-ftdi#chip-notes) on a CSR chip? Did You powered bluetooth board?

Limoster, 2016-11-14:
It's a generic SANWU Audio TDA7492P 25W+25W amplifier and bluetooth board.
Version that has 3.5mm jack audio input (board version 3.2; version 3.3
does not have it).
It's incredibly well designed and it can drive 15W RMS without heatinsink.
No distorsion (>1%)until 15W @12V. No background digital noise.

Under bluetooth board there are 7 solder pads that are used for
programming. The pinout is in the posts above.

d0narnbl1tz, 2016-11-14:
@lorf 
It´s using an csr8635 as bluetooth chip, i think uhm the only unconnected port that is associated with spi would be spipcm, didnt mentioned that pin as an enable 
when i come home this evening or maybe tomorrow i will look what happens with him high.

About powering, yeah, but i keept it simple, used the onboard supplay of the whole board

@Limoster 
its the version without 3,5mm jack, maybe when i have to much time...
yes its very well designed, wanted just a cheap solution for building one that will work under the circumstances i want and got something thats partly better than some expensive ones^^

Limoster, 2016-11-15:
First you should grab a multi-meter with the tester function selected and check each of the 7 pads from left to right (the button is on the bottom), and solder the following connections to FT232 Breakout Board: 
[image]

---Bluetooth-------FT232-----
1- SPI_PCM  ---> 3.3V With 10Kohm resistor
2- GND --->        GND
3- VBAT --->       3.3V
4- MOSI --->       RI
5- CLK --->          RTS
6- CSB --->         DTR
7- MISO --->       DSR (or RSD if you have counterfeit board)
----------------------------------

Amplfier/bluetooth board has to be powered on. FT232 can't supply enough current to the module by itself.