lorf/csr-spi-ftdi #17: Where can I get CSR DSP Equalizer configurator?

Limoster, 2017-02-13:
I'm trying to configure the internal Kalimba DSP in CSR8635 but I can't seem to find the software to configure it. The datasheet says it works with CSR8600 Rom Series configurator.
Does anyone have a copy or knows where I can get one in my hands?

lorf, 2017-02-13:
You, probably, need a tool called "Parameter manager", it is contained in the CSR "Universal Front End" (UFE) package. Searching for "CSR8600 ROM Series A03 UFE" reveals some links.

Limoster, 2017-02-13:
I'll test it when I get my hands on my new amplifier bluetooth combo.

I'm now doing my bachelor's research project. I'm doing a "R&D of a
portable speaker".

Many thanks!

Limoster, 2017-02-15:
I've just downloaded the programme from btspeaker.cn and it looks like the thing I wanted.
I wondered what wpuld happen If i searched for A04 version and I found the CSR8635 datasheet.
According to it, the A04 is the ROM for CSR8635, and the CSR8645 needs the A03.

The only sites I have found the utility A04 are down, but in the name of the file it says "_A03_A04_"

Limoster, 2017-04-26:
I'm having a problem.
I'm getting an error when I try to connect the FT232RL to the A03 Tool. I have changed the usbspi.dll so it recognizes the device but it still doesn't let me change the variables.
Is it because this programmer is meant to work only with the Development Board?

Limoster, 2017-04-26:
I have found this:
http://g01.s.alicdn.com/kf/HTB11F5VFVXXXXbxXXXXq6xXFXXXj/220726009/HTB11F5VFVXXXXbxXXXXq6xXFXXXj.jpg
In the top left corner you can se a USB connection to the main board. Is this the way the A03 connects to the chip?

VincentGijsen, 2017-05-27:
You should definitely check-out https://www.dropbox.com/s/vtvg8g7qnk08pju/CSR86XX%20ConfigurationTool-3.0.70.zip?dl=0 the version 3.0.70 is quite reliable when it comes to directly updateding the chip, and keeps licences in place... little issues with this one opposted to the >3.0.70 versions (using csr8645 btw). The Equalizer tool, howerver doens't recognize the ftdi programmer unfortunately (https://www.dropbox.com/sh/11zmhypve7m986a/AADbcGpB9pltSB547kKDAsFga?dl=0) has the universal parameters program). But i'm unable to get it to connect via the ftdi :(, tried windows 10, and windows 7 32bit (in oracle vm).... Realy currious as to your results!

timkausw, 2017-07-10:
Get anyone the UniversalFrontEnd.exe running with the ftdi? I did a bit of research and compared the original usbspi.dll with the precompiled for windows from this project. I found out that two functions are not implemented they are "spifns_stream_bccmd_init" and "spifns_stream_bccmd_cmd". It would be great if the UniversalFrontEnd.exe will run to adjust the equalizer.

VincentGijsen, 2017-07-10:
I gave up on the UniversalParameter tool in conjunction with the FTDI-SPI.

I bought a chinese CSR-SPI unit (18Euro) , and after some juggling with the
wires (which for some reasone are not equally cloned), it works  reasonably
stable (with an occational crash..

Never got any luck with the FTDI-implementation though.

timkausw, 2017-07-11:
:) i started with the Chinese csr programmer and it worked with the PSTool round about 2 hours. As I tried the Universal Parameter Tool it was already burned :( so I decided to try the fdti way with hope of more reliability of the hardware. I hope "lorf" will upgrade, please!

maik-dmxc, 2017-11-23:
dear lipton5001
is it possible to get connection information from the bluetooth modul via a serial bus?
like, name off connected phone or connection status?

lipton5001, 2017-11-23:
@maik-dmxc

I really don't know, given how capable these chips are, I am 100% sure something of this nature is achievable (cars and such would definitely require this so I highly doubt CSR would leave t out)

Go through the extended datasheets of the CSR8645, and if that doesn't have it, check the CSRA64215, that's the only stronger chip I know of