lorf/csr-spi-ftdi #19: Invalid Control Data on any chip.

AM00, 2017-03-12:
Hello, I have problem, I try to re-program CSR8645 and HC05 and nether of these work. I try on XP, 7 and 10 but it has no effect. Always got error like this:  Unable to start read (Invalid Control Data)

lorf, 2017-03-12:
"Invalid Control Data" errors indicate the failure to establish SPI communication to the target chip. In absence of any details I can only point to the [Troubleshooting section in the README](https://github.com/lorf/csr-spi-ftdi#troubleshooting).

AM00, 2017-03-12:
I already tried to do everything in troubleshooting section. Here are some more details:
- When using version 2.5 (BlueSuite) error is Invalid Control Data. On 2.4 i Could not communicate with chip. 
- I tried to use 220/100/No resistors.
- Tried different version of drivers.
I think it may be FTDI fault but it programmed Arduino and ESP8266 without any trouble.

lorf, 2017-03-12:
What FTDI chip, what FTDI board do You use? Is this counterfeit chip (costing < $5)? What revision of FTDI chip (it's on top of the chip, something like "1234-C")?

Does your bluetooth module powered from separate power supply?

AM00, 2017-03-12:
Yup, it is counterfeit (1.58$, have it for around half year) and it has 1412-G labeled. Looks like ordinary FTDI board.

AM00, 2017-03-12:
And I use LM2596S based step-down module with ground hooked to FTDI. Using FTDI as power supply also changes nothing.

lorf, 2017-03-13:
Can You test with blueflashcmd and show the output?

```
BlueFlashCmd.exe -chipver
BlueFlashCmd.exe -identify
```

AM00, 2017-03-14:
It all looks the same. Even if chip is not connected. 

C:\Program Files (x86)\CSR\BlueSuite 2.5.0>BlueFlashCmd.exe -chipver
blueflashcmd, version 2.5.0.93 Release
Copyright (C) 2002-2012, Cambridge Silicon Radio Ltd.

22:48:30.910203: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.2, git rev 4c3061a
22:48:31.114886: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.118026: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.121258: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.125148: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.128212: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.131192: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.134165: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.137155: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.140153: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.144113: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.147136: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.150475: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.155071: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.159011: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.161369: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.166026: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.168992: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.171964: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.174945: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.177899: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.181862: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.185901: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.189750: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.193786: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.196605: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.199473: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.203510: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.207573: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.211599: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.213886: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.219676: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.223643: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.225829: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.229836: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.232839: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.237544: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.240494: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.242968: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.247509: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.251440: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.253859: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.259443: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.262372: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.265394: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.271341: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.274348: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.280070: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.283925: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.288108: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.289419: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
22:48:31.295016: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
[... cut ...]
FTDI stats: 1.#R xfers/s (103025154595274140000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.00 short reads/s,
            134 xfers/4 short reads in 0.00 s,
            1.00 xfers/IO, 82.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Failed

C:\Program Files (x86)\CSR\BlueSuite 2.5.0>

lorf, 2017-03-17:
Sorry for a delay. I, currently, have no ideas other than FTDI adapter fault.

AM00, 2017-03-18:
Yup, that's what i was thinking.

tomaskovacik, 2017-05-04:
same here, problem was I was using 5V from ftdi board to power bluetooth board, after powering it from diferent USB via diod (to make it closer to 4.2V) everything works

devitalii, 2017-05-15:
I thought this error was my FTDI ($1.5 ebay) problem too, BUT IT WAS THE PROBLEM with my connection to the HC05/06 chip's ISP pins (CLK, MO,MI, CSB) !!!! Moved the wires and all works Great! (more info - email me)

err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)

tomaskovacik, 2017-05-16:
true, also in datasheet provided by sellers of CSR8645 on ebay is CS# pin market as CSB, on this site, it's marked SPI_CS#

CS# 	        2 	DTR# 	D4 	CSB#
CLK   	3 	RTS# 	D2 	SPI_CLK
MOSI 	6 	RI# 	        D7 	SPI_MOSI
MISO 	9 	DSR# 	D5 	SPI_MISO

pabolojo, 2017-08-11:
@devitalii i am having some problems too. I have wired everything well with 100 Ohm and 240 Ohm resistors and It always shows the same error: "Error detecting chip type (Unable to start read (invalid control data))", also i am powering the hc05 with an external supply voltage.
Any suggestions?

devitalii, 2017-08-11:
!!!!!!!!!!! Here are my process notes !!!!!!!!!!!!, might help. 

Basically, important to make sure connection is rigid! 

Also use right drivers, right BlueSuite and .dll file  (see the notes file with images):

https://www.dropbox.com/s/k033b2y0q8gsz89/%21%21%21%21%21%21%21%21%21%21%21%21%21%21%21%21%21%21%21README--HC05flash--shared%20Notes%21.rtf?dl=0

snackfart, 2019-08-12:
on ftdi boards dsr == rso