lorf/csr-spi-ftdi #23: Bluecore can't connect to CSRA64215

kingmercian, 2017-07-29:
Hi @lorf , I am hoping you can tell me where I am going wrong!

Connections:

| CSRA64215 | Resistor  | FTDI  | BATT  |
|:-:|:-:|:-:|:-:|
| GND  |   |   | GND  |
| 3.3V  |   |   |  3.3V |
|  SPI_EN | 10KΩ  |   | 3.3V |
| SPI_MOSI 220KΩ |   |  RI |   |
| SPI_CLK  |  220KΩ |  RTS |   |
|  SPI_CSB | 220KΩ  | DTR  |   |
| SPI_MISO |  220KΩ |  DSR |   |

I am using BlueSuite 2.6.x and my env is Windows 10, I am using the generic fake FTDI with pinout from eBay and I am using libusbK with your .dll replacing the one within BlueSuite.

BlueSuite detects the FTDI interface but when attempting to connect I get "Error detecting chip type (unable to start read (invalid control data))"

Any input would be appreciated.

kingmercian, 2017-07-29:
After more digging:

> C:\Program Files (x86)\CSR\BlueSuite 2.6.2>BlueFlashCmd.exe -chipver
blueflashcmd, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2002 - 2015.

>13:09:37.500194: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.2, git rev 4c3061a
13:09:37.525193: warn:spi.c:497:spi_enumerate_ports: FTDI: ftdi_usb_get_strings() failed: [-7] libusb_get_string_descriptor_ascii() failed
13:09:37.610222: err:spi.c:728:spi_open: FTDI: ftdi_usb_open_desc() failed: [-6] ftdi_usb_reset failed
==============================================================================
ERROR: Couldn't find SPI port
==============================================================================
Failed

So looks like it isn;t connecting to the FTDI even though it all looks good from device manager:
[image]

kingmercian, 2017-07-29:
Also looks similar to [this](https://github.com/lorf/csr-spi-ftdi/issues/7) issue, as that user did I also use two PSUs (one to power FTDI and one for CSRA64215) and I had also tried connecting ground from FTDI to CSRA64215 but this causes the FTDIs LEDs to light up permanently and prevents it from connecting to the computer.

lorf, 2017-07-29:
I think 220 KOhm in signal lines is too much. Please try 220 Ohm resistors.

kingmercian, 2017-07-29:
My bad, I meant 220 Ohm.

For some reason it started working after unplugging and replugging the ft232 a few times.

Thank you.

closed 2017-07-29 by kingmercian