lorf/csr-spi-ftdi #25: Error with RN421 firmware on HC-05 module with csr-spi-fdti driver

romandesign, 2017-12-22:
I have a problem, please advise: I have exactly an HC-05 board from Ali Express, which I want to upgrade to RN42 firmware. I got connected to HC-05 (LED blinking when "processor running"), I dumped backup firmware and settings, uploaded RN42 firmware, clicked "Start Processor" and got "Processor Running" but LED doesn't blink, it's off. When I try to connect with PSTool I get an error: "unable to find entries in the look-up table on chip". When I flash backup firmware back everything works again, but when I flash RN-42 - same thing, can't get to settings, LED not blinking, but can start and stop processor, get firmware version, verify etc. Any advice?﻿ Is it driver-related or hardware-related? Anyone had similar issue? Thanks.

doteroargentino, 2018-01-13:
I have this problem aswell, any news?

romandesign, 2018-01-14:
I can confirm that my module that failed is the one that appears on that photo as the bad one on the bottom. However in the product photo on Ali Express the product shows the top one. So it's difficult to get that one I think. I have ordered the one form a link to one of the videos a while ago, let's hope I get the right one in the mail...  It was called hc-06

ssapalski, 2018-01-14:
I can confirm that the module shown as the bad one on the photo doesn't work with RN-42 firmware. Reverting the module to the original firmware brings it back to life!

```
$ e2cmd info
Chip ID - 0x4543
Chip Name - BC3 MM (kal)
Unable to calculate addressing mode of EEPROM
```

```
$ BlueFlashCmd identify
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00c2, dev_id = 0x225b
Firmware ID (loader)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Firmware ID (stack)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Result: Usable flash size: 128 sectors, 8 megabit.
```

Currently I can't compare this with a working unit but this looks like a BC3 and not BC4 module. Might be that it was just labeled BC4!

witnessmenow, 2018-01-14:
@ssapalski if there are any commands or tests you want me to run I'd be happy to do it. I have both types here and also have the ftdi circuit set up

@romandesign I mentioned in my pinned YouTube comment, I bought 4 modules off 4 different sellers, all looked like the good module on the listing but two of the ones I received were the bad ones. The one I linked to in the video is the only one I know for sure what seller I got it from (and was good)

It was also bought in august 2016 so maybe its old stock

doteroargentino, 2018-01-14:
@witnessmenow that is because most sellers have outdated pictures on their listings but they say they sell the most recent version of the module, maybe manufacturers realized people were turning this modules into HID's and decided to screw us up?

witnessmenow, 2018-01-14:
Maybe, I bought 3 of the modules in September 2017 and one of them was good, but I have no way of knowing which seller sent which one!

I'd say someone just figured how to make a hc05 module cheaper! I'm sure sellers don't really care what we do with them.

doteroargentino, 2018-01-14:
I think they do care, since they use the same chips for both modules I'd assume they wouldn't want us to have an HID module for the price of an HC-05, there's a huge price gap between them

ssapalski, 2018-01-14:
@witnessmenow 
Can you run the two commands I've shown on the good unit?

**e2cmd.exe info
BlueFlashCmd.exe identify**

My assumption is that this will show a BC4 device (BlueCore 4)!
If this is true, than it would explain why the RN-42 firmware isn't running since most likely this firmware can only run on BC4 devices. These are just some assumptions, I'm playing around with these BT devices since a couple of days and I don't know them well.

witnessmenow, 2018-01-14:
Good board (already running RN-42 can revert to stock if needed)

`C:\Program Files (x86)\CSR\BlueSuite 2.6.2>e2cmd.exe info
e2cmd.exe, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2007 - 2015.

21:32:27.171178: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.2, git rev 4c3061a
Chip ID - 0x4826
Chip Name - BC4-EXT (cyt)
21:32:29.420610: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
Unable to calculate addressing mode of EEPROM
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.2 (git rev 4c3061a)
Time open: 2.86 s
Time in xfer: 2.60 s (90.86% of open time)
Reads: 224 (13372 bytes, 59.70 bytes avg read size)
Writes: 320 (15084 bytes, 47.14 bytes avg write size)
Xfer data rate: 10.67 KB/s (28456 bytes in 2.60 s)
IOPS: 208.00 IO/s (544 IOs in 2.60 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1085.00 xfers/s (0.00 short reads/s,
            2829 xfers/1 short reads in 2.60 s,
            5.00 xfers/IO, 323.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************

C:\Program Files (x86)\CSR\BlueSuite 2.6.2>BlueFlashCmd.exe identify
blueflashcmd, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2002 - 2015.

21:33:06.033075: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.2, git rev 4c3061a
Resetting XAP
Identifying XAP
21:33:07.012860: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:33:07.040974: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00c2, dev_id = 0x225b
Firmware ID (loader)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Firmware ID (stack)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Result: Usable flash size: 128 sectors, 8 megabit.
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.2 (git rev 4c3061a)
Time open: 4.13 s
Time in xfer: 3.81 s (92.17% of open time)
Reads: 399 (20628 bytes, 51.70 bytes avg read size)
Writes: 560 (20177 bytes, 36.03 bytes avg write size)
Xfer data rate: 10.45 KB/s (40805 bytes in 3.81 s)
IOPS: 251.00 IO/s (959 IOs in 3.81 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1093.00 xfers/s (0.00 short reads/s,
            4169 xfers/1 short reads in 3.81 s,
            4.00 xfers/IO, 314.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************
Success`

Bad board (stock)

`C:\Program Files (x86)\CSR\BlueSuite 2.6.2>e2cmd.exe info
e2cmd.exe, version 2.6.2.632 Release
Copyright Cambridge Silicon Radio Limited 2007 - 2015.

21:38:45.697115: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.2, git rev 4c3061a
21:38:46.495925: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.498125: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.502984: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.507925: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.510845: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.513974: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.519845: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:46.523779: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
[... 66 lines cut ...]
21:38:59.933685: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:59.934799: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
21:38:59.940685: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
Resetting XAP
21:39:00.045309: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
Identifying XAP
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00c2, dev_id = 0x225b
Firmware ID (loader)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Firmware ID (stack)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Result: Usable flash size: 128 sectors, 8 megabit.
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.2 (git rev 4c3061a)
Time open: 12.63 s
Time in xfer: 12.18 s (96.48% of open time)
Reads: 450 (70360 bytes, 156.36 bytes avg read size)
Writes: 613 (69963 bytes, 114.13 bytes avg write size)
Xfer data rate: 11.24 KB/s (140323 bytes in 12.18 s)
IOPS: 87.00 IO/s (1063 IOs in 12.18 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1032.00 xfers/s (0.00 short reads/s,
            12589 xfers/3 short reads in 12.18 s,
            11.00 xfers/IO, 357.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Success`

ssapalski, 2018-01-15:
@witnessmenow 
Many thanks, this looks like my assumption was correct. I will compress the information to the relevant parts:

**good board (rn-42 firmware):**
```
$ e2cmd.exe info
Chip ID - 0x4826
Chip Name - BC4-EXT (cyt)
Unable to calculate addressing mode of EEPROM

$ BlueFlashCmd.exe identify
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00c2, dev_id = 0x225b
Firmware ID (loader)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Firmware ID (stack)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Result: Usable flash size: 128 sectors, 8 megabit.
```

**bad board (stock firmware):**
```
$ e2cmd.exe info
Chip ID - 0x4543
Chip Name - BC3 MM (kal)
Unable to calculate addressing mode of EEPROM

$ BlueFlashCmd.exe identify
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00c2, dev_id = 0x225b
Firmware ID (loader)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Firmware ID (stack)="bc3k_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Result: Usable flash size: 128 sectors, 8 megabit.
```

Can you please do the same with the "good board" and the stock firmware? I guess this won't change anything related to "chip id" and "chip name". If the "chip id" value is unrelated to the running firmware, than this would mean that in fact the "bad board" is a BC3 module but fake labeled BC4.

Baton34, 2018-02-03:
**good board (stock firmware):**
```
>e2cmd.exe info
Chip ID - 0x4826
Chip Name - BC4-EXT (cyt)
Unable to calculate addressing mode of EEPROM

>BlueFlashCmd.exe identify
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x007f, dev_id = 0x225b
Firmware ID (loader)="cyt_8unified_fl_bt2.0_22_0612121241_encr56 2006-12-12"
Firmware ID (stack)="cyt_8unified_fl_bt2.1_23g_0903311011_encr56 2009-03-31"
Result: Usable flash size: 128 sectors, 8 megabit.
```

**good board (rn-42 firmware):**
```
>e2cmd.exe info
Chip ID - 0x4826
Chip Name - BC4-EXT (cyt)
Unable to calculate addressing mode of EEPROM

>BlueFlashCmd.exe identify
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x007f, dev_id = 0x225b
Firmware ID (loader)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Firmware ID (stack)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Result: Usable flash size: 128 sectors, 8 megabit.
```

token0, 2018-02-08:
BC417143B part is originally BlueCore4 (BC4). Looks like some smartasses repackaged BlueCore3 chip into 7x7 mm WFBGA (originally 10x10 mm 96-ball LFBGA) and labeled as BlueCore4. Original BlueCore4 is only 8x8mm 96-ball TFBGA and 6x6mm 96-ball VFBGA package options. So that's definitely counterfeit. BlueCore3 is old as dump of mammoth (2003).

[image]

ssapalski, 2018-02-08:
Yes, these are counterfeit chips with BC3 compatible HC-05 firmware and thus not usable with RN42 (BC4) firmware. The csr-spi-fdti driver works fine with these chips!

witnessmenow, 2018-02-08:
Probably worth closing so, not a driver issue

lorf, 2018-03-28:
Let it hang here for information purposes.

Gunvarrel39, 2018-03-29:
Hi! I am currently facing this problem right now, unable to access the PSTool after flashing RN42's firmware into my HC05 (LED stops blinking). As per witnessmenow's photo, it seems that my HC05 is indeed the 'bad' one (bottom on picture). So I can't continue this project unless I get the 'good' one or is there any other way to do it? I've bought two HC05 modules and both of them is the 'bad' one. :(

Help would be very much appreciated, thanks!

doteroargentino, 2018-03-29:
@Gunvarrel39 if you'd read all the replies you'd know that the 'bad' modules have a BlueCore 3 chip that can't be flashed with RN42's firmware so no, you need to get a 'good' one if you want to flash it.

jmvermeulen, 2018-11-12:
Personally I bought this one:
https://www.aliexpress.com/item/HC05-HC-05-master-slave-6pin-JY-MCU-anti-reverse-integrated-Bluetooth-serial-pass-through-module/32340945238.html

Didn't test the custom firmware but the unit chips are the same as the 'working' variant on the picture.
Order was from august this year.

bilogic, 2018-11-29:
I think the working one is sold as HC-06 while the non-working is sold as HC-05

bilogic, 2019-01-15:
I'm not sure what you guys plan to do with this HC-05, but the ESP32 supports BLE.. it may be an easier and more readily available alternative if you are don't plan on supporting Bluetooth 2.0 etc.

davejel, 2019-01-15:
Hi thanks for the reply
RN-42 firmware allows the use as a hid keyboard or mouse, it is possible to replace the firmware on hc-05/06.
Cost of RN-42 is about £20. If they can be re-programmed it means HC-05/06 will do the same job for a quarter of the price
Manufacturers have found out about this and in some cases changed the boards so that they will not work as
RN-42 , hence my question to you
I have since done more research and I think your boards can be used as RN42 so I have ordered your last 3 so that I can try it out, if you are interested I can come back to you and let you know for certain

I’ll have a look at the ESP32 and see if it will do the same job, thanks

Cheers
Dave

davejel, 2019-01-16:
hi @bilogic my goal is to 'create' a simple BT switch which can then be used by disabled users to use programs that need input.
the switch would be configured as a hid keyboard with only one key
some HC-06 boards can be uploaded with RN42 firmware, RN42 is naturally a Hhid
at command set (which is then needed for configuration) can be done by connecting the HC to an ESP8266 via serial
if you pm me we can compare notes and work together if there is enough common ground

bilogic, 2019-01-18:
@davejel i think https://github.com/asterics/esp32_mouse_keyboard will almost work right out of the box for you, just add a button to one of the IO pins. I was able to compile and get it to work which is why I have the confidence to move forward with my project, however my main challenge eventually will be in drawing the PCB.

davejel, 2019-01-18:
@bilogic looks interesting, i have just bought some HC-06 chips which i shall be testing today(i think they are the ones which will take RN42 software) i'll report back here later and let you know how i get on with it.

looks as though the approach you are taking might be better in that there is no need for RN42 firmware to make the hid.
i have been following https://mitxela.com/projects/bluetooth_hid_gamepad
and https://www.youtube.com/watch?v=y8PcNbAA6AQ
both these use HC-05/06 with RN-42 firmware programmed into them
My final aim is to produce a bluetooth switch for use by disabled people at a reasonable price at the moment the cheapest one on the market is £175 which most cannot afford, if i can produce one for say £25 - £30 or under this could potentially help a lot of disabled users.
i dont mind how this is achieved  so i will buy some ESP-32 and try them as well to see how it all pans out
Dave

wimmatthijs, 2019-03-07:
Thanks for the info, i ordered 3 modules online and got 3 bad ones... how unfortunate.
Does anybody know if we can still ordere the old style HC05/06 ?
Thanks

adamhun94, 2019-03-28:
@davejel Can you confirm if your module is working or not? I can't find any working modules right now.. If anyone has a working one, please share the link with us.

assasinmaxy, 2019-04-07:
I was faced with the same problem. If a decision on HC-05 appears, please let me know.

LouDnl, 2019-11-23:
So I know this issue talks about th BC4 and BC3. But I am facing the same problem with an HC-05 module that identifies as BC5. Do I also have a fake module?

e2cmd:
20:29:41.415857: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
Chip ID - 0x50e1
Chip Name - BC5-MM (elv)
Unable to calculate addressing mode of EEPROM
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 6.84 s
Time in xfer: 6.54 s (95.59% of open time)
Reads: 200 (44266 bytes, 221.33 bytes avg read size)
Writes: 311 (44939 bytes, 144.50 bytes avg write size)
Xfer data rate: 13.31 KB/s (89205 bytes in 6.54 s)
IOPS: 78.00 IO/s (511 IOs in 6.54 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1188.00 xfers/s (0.00 short reads/s,
            7780 xfers/2 short reads in 6.54 s,
            15.00 xfers/IO, 367.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************
BlueFlashCmd:
20:30:02.511466: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
Resetting XAP
Identifying XAP
Flash identity: size = 128 sectors (8 Mbit), man_id = 0x00bf, dev_id = 0x273e
Firmware ID (loader)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Firmware ID (stack)="cyt_8unified_fl_bt3.0_23i_1002111152_native_encr56 2010-02-11"
Result: Usable flash size: 128 sectors, 8 megabit.
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 11.51 s
Time in xfer: 11.09 s (96.40% of open time)
Reads: 377 (78984 bytes, 209.51 bytes avg read size)
Writes: 561 (78563 bytes, 140.04 bytes avg write size)
Xfer data rate: 13.86 KB/s (157547 bytes in 11.09 s)
IOPS: 84.00 IO/s (938 IOs in 11.09 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1240.00 xfers/s (0.00 short reads/s,
            13767 xfers/1 short reads in 11.09 s,
            14.00 xfers/IO, 366.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************
Success

darkautism, 2020-03-04:
Hello everyone,

If you get a fake HC-05 or HC-06 with BC3-MM(kal) chip, and you just wanna use your HC-05/HC-06 to pair with your Dualshock 4 controller. You can use a custom firmware which i wrote for BC3-MM(kal).

This is custom firmware and source code is here:
https://github.com/darkautism/HC-05-Dev

And i think it is possible write a custom firmware for HC-05/HC-06 as a dongle.

Himanshuj123, 2020-07-16:
Hello everyone, i also got 2 bad chips (BC-3) and I got the same error in pstool(unable to load entries) but when I goona to flash rn-42 firmware on it the blueflash warnings me that the (image  size is larger then flash size, image get truncated) 
and after inspecting that the flash size of these fake modules is 6mb but the rn-42  image size is 6.13mb so that's why the pstool looks for the settings in the flash which it wouldn't get as the image get compressed while flashing

SeungheonOh, 2021-02-25:
So, I can conclude the problem is: some board using BC3 which is NOT compatible with RN-42. Does that mean we can generalize any HC-05 with CSR BC417143 is good and compatible? 

Is there any impact of the 2nd chip on the lower board? I'm seeing multiple variations: BA1237, cFeon EN29L, and other chip by "MX".

SeungheonOh, 2021-02-25:
[image]
My chip really have weird layout: Crystal on the other side, unusual label (R201203M), and etc.

spambake, 2021-12-24:
I have a module very similar to SeungheonOh's.  The regulator on the carrier board is actually 3.6v not 3.3v.  Removing and bypassing the regulator shows the module will not operate at 3.3v but needs 3.6v.  Some kind of super cheap fake part I guess.

BastelPichi, 2022-01-21:
So... I tried this too.  So here's a list of what I got and how I fixed (most) of them.
 - Error Message: Unknown Side by Side Configuration by Windows.
    Description: During the Installation of BlueSuite (2.5 and 2.6.6 )it always failed to install the .NET framework.
    Fix: Download and Install: https://www.microsoft.com/en-us/download/confirmation.aspx?id=6041.
 - PSTool communication Error
   Fix: Just hit retry. First time always failed for me.

For the Faked Chips, well...
For my first try of this a 2 years ago, I ordered some hc05 chips on banggood. (Link below) I couldnt get them to work, always got the SPI error in BlueFlash.
If it was my bad soldering back then, or some other thing that destroyed these, id dont know. However, the chips on the Board are an BC417!

Two days ago, ordered another Pair. They worked fine, but they have the typical BC352. So they wouldn't work with the new Firmware. The where advertised as an "Master Slave Module"

While looking trough my Pile of Arduino related Parts, found a Seed Bluetooth shield. It uses the BC417 and I will try to flash it know.

Have attached Photos of my Modules down below, will make some better tomorrow.
[image]
[image]

Link to what was an BC417: https://www.banggood.com/Geekcreit-HC-05-Wireless-bluetooth-Serial-Transceiver-Module-Slave-And-Master-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-908621.html?rmmds=myorder&cur_warehouse=CN
However that one says BC352 on the Photos...

BastelPichi, 2022-01-21:
the shield flash seems to work.

alexeygolev, 2023-03-27:
@SeungheonOh I have the same version that you have. Have you managed to figure out how to work with it?

BastelPichi, 2023-03-27:
> @SeungheonOh I have the same version that you have. Have you managed to figure out how to work with it?

Just use an esp32 🙃

SeungheonOh, 2023-03-27:
yep, I returned it and got esp32

assasinmaxy, 2023-03-28:
> > @SeungheonOh I have the same version that you have. Have you managed to figure out how to work with it?
> 
> Just use an esp32 🙃

the same )

BastelPichi, 2023-03-28:
This is a great project to learn about firmware, reflashing and similar, but if you just want an soloution, just go with the esp32. Its just a few bucks...

OhhhThatVarun, 2024-10-25:
@darkautism, is it impossible to write a firmware that makes "bad" chips HID-compatible?

viniciusvillas, 2025-07-04:
Sorry to resurrect this thread, but this left me without words. The fake modules are kinda incredible.
This one had no memory chip at all. The memory "chip" was just a placeholder with 16 pins connected to ground.

[image]
[image]
[image]
[image]