lorf/csr-spi-ftdi #27: Changing the Pinout

sirtet, 2018-01-20:
I don't have a breakout board for the FTDI, but a sparkfun FTDI Basic (USB-Serial) Board.
I don't dare to solder on the chip directly, too tight space.

Would it be possible to change the pin-config to what's used there?
Pins 1,2,5,11
https://cdn.sparkfun.com/datasheets/BreakoutBoards/FTDI%20Basic-v22-3.3V.pdf

PS:
Would it not be a nice idea to have the pinout in a config file, that can be changed after compilation?

lorf, 2018-03-27:
Added `sparkfun` pinout to the devel branch. Here are precompiled binaries for testing: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a3 .

lorf, 2018-03-28:
Pinout is here: https://github.com/lorf/csr-spi-ftdi/blob/devel/README.md#options

commit 48223fe7 (2018-04-09): Add pinout for Sparkfun FTDI Basic

Fixes #27