lorf/csr-spi-ftdi #28: Unhandeld exception when communicating with HC05

teeheee, 2018-02-09:
Hello,

I tried this code a year ago and it worked good. Now I wanted to use it again with the same setup but it always breaks when Blueflash communicates with the Module. I tried it on two different PCs with wine-1.6.2, windows7 and windows10 and tried the prebuild version and build it myself. It is always the same behaviour. I also tried different usb Ports. 

TLDR:
OS:          Xubuntu with wine-1.6.2 or windows7 or windows10
Bluesuite: 2.4
usbspi.dll: prebuild-0.5.1, prebuild-0.5.2 and ownbuild-0.5.2 (for all OSs)
FTDI-CHIP: probably real FT232RL
Problem: Bluesuite crashes when using the usbspi.dll.

Here is the stacktrace I got from wine when using the prebuild version of the dll. 

>Unhandled exception: page fault on read access to 0x45455246 in 32-bit code (0x7e089124).
Register dump:
 CS:0023 SS:002b DS:002b ES:002b FS:0063 GS:006b
 EIP:7e089124 ESP:0033d4fc EBP:0033d544 EFLAGS:00010202(  R- --  I   - - - )
 EAX:45455246 EBX:7e098000 ECX:798ef768 EDX:00000000
 ESI:0041a5c0 EDI:00000001
Stack dump:
0x0033d4fc:  7e0938f6 7e092f89 0000016d 00000004
0x0033d50c:  7e0930c3 00000001 00000010 45455246
0x0033d51c:  0000d6c0 00000000 5a7d53b4 00010b9d
0x0033d52c:  00000000 00000000 00000000 00000000
0x0033d53c:  7e098000 7e098000 0033d594 7e087897
0x0033d54c:  00000001 00000010 45455246 0000d6c0
Backtrace:
=>0 0x7e089124 spi_xfer+0xfe() in usbspi (0x0033d544)
  1 0x7e087897 in usbspi (+0x7896) (0x0033d594)
  2 0x7e087fcc spifns_stream_sequence+0xe2() in usbspi (0x0033d5d4)
0x7e089124 spi_xfer+0xfe in usbspi: movzwl	0x0(%eax),%eax
Modules:
Module	Address			Debug info	Name (65 modules)
PE	  340000-  35f000	Export          pttransport
PE	  400000-  412000	Deferred        blueflash
PE	10000000-10023000	Deferred        flash
PE	5d360000-5d370000	Deferred        mfc80deu
PE	78130000-781cb000	Deferred        msvcr80
PE	781d0000-782e0000	Deferred        mfc80
ELF	7b800000-7ba54000	Deferred        kernel32<elf>
  \-PE	7b810000-7ba54000	\               kernel32
ELF	7bc00000-7bcda000	Deferred        ntdll<elf>
  \-PE	7bc10000-7bcda000	\               ntdll
ELF	7bf00000-7bf04000	Deferred        <wine-loader>
PE	7c420000-7c4a7000	Deferred        msvcp80
ELF	7e071000-7e09d000	Dwarf           usbspi<elf>
  \-PE	7e080000-7e09d000	\               usbspi
ELF	7e09d000-7e192000	Deferred        comctl32<elf>
  \-PE	7e0a0000-7e192000	\               comctl32
ELF	7e1ec000-7e22d000	Deferred        usp10<elf>
  \-PE	7e1f0000-7e22d000	\               usp10
ELF	7e278000-7e2ad000	Deferred        uxtheme<elf>
  \-PE	7e280000-7e2ad000	\               uxtheme
ELF	7e2ad000-7e2b4000	Deferred        libxfixes.so.3
ELF	7e2b4000-7e2c0000	Deferred        libxcursor.so.1
ELF	7e2c0000-7e2d3000	Deferred        libxi.so.6
ELF	7e2d3000-7e2d7000	Deferred        libxcomposite.so.1
ELF	7e2d7000-7e2e4000	Deferred        libxrandr.so.2
ELF	7e2e4000-7e2f0000	Deferred        libxrender.so.1
ELF	7e2f0000-7e2f7000	Deferred        libxxf86vm.so.1
ELF	7e2f7000-7e2fb000	Deferred        libxinerama.so.1
ELF	7e2fb000-7e302000	Deferred        libxdmcp.so.6
ELF	7e302000-7e306000	Deferred        libxau.so.6
ELF	7e306000-7e32c000	Deferred        libxcb.so.1
ELF	7e32c000-7e477000	Deferred        libx11.so.6
ELF	7e477000-7e48c000	Deferred        libxext.so.6
ELF	7e48e000-7e4b2000	Deferred        imm32<elf>
  \-PE	7e490000-7e4b2000	\               imm32
ELF	7e4b4000-7e541000	Deferred        winex11<elf>
  \-PE	7e4c0000-7e541000	\               winex11
ELF	7e669000-7e693000	Deferred        libexpat.so.1
ELF	7e693000-7e6dc000	Deferred        libfontconfig.so.1
ELF	7e6dc000-7e707000	Deferred        libpng12.so.0
ELF	7e707000-7e722000	Deferred        libz.so.1
ELF	7e722000-7e7d2000	Deferred        libfreetype.so.6
ELF	7e7fa000-7e870000	Deferred        shlwapi<elf>
  \-PE	7e810000-7e870000	\               shlwapi
ELF	7e870000-7e91f000	Deferred        msvcrt<elf>
  \-PE	7e880000-7e91f000	\               msvcrt
ELF	7e945000-7ea5c000	Deferred        gdi32<elf>
  \-PE	7e950000-7ea5c000	\               gdi32
ELF	7ea5c000-7ebaa000	Deferred        user32<elf>
  \-PE	7ea70000-7ebaa000	\               user32
ELF	7ebaa000-7ec16000	Deferred        advapi32<elf>
  \-PE	7ebc0000-7ec16000	\               advapi32
ELF	7ec16000-7ec29000	Deferred        libnss_files.so.2
ELF	7ec29000-7ec36000	Deferred        libnss_nis.so.2
ELF	7ec36000-7ec51000	Deferred        libnsl.so.1
ELF	7ef83000-7efd8000	Deferred        libm.so.6
ELF	7efe7000-7f000000	Deferred        version<elf>
  \-PE	7eff0000-7f000000	\               version
ELF	f7355000-f735f000	Deferred        libnss_compat.so.2
ELF	f7361000-f7366000	Deferred        libdl.so.2
ELF	f7366000-f751c000	Deferred        libc.so.6
ELF	f751c000-f7539000	Deferred        libpthread.so.0
ELF	f7561000-f7716000	Dwarf           libwine.so.1
ELF	f7717000-f773c000	Deferred        ld-linux.so.2
ELF	f773f000-f7740000	Deferred        [vdso].so
Threads:
process  tid      prio (all id:s are in hex)
0000000e services.exe
	0000001e    0
	0000001d    0
	00000014    0
	00000010    0
	0000000f    0
00000012 winedevice.exe
	0000001c    0
	00000019    0
	00000017    0
	00000013    0
0000001a plugplay.exe
	00000020    0
	0000001f    0
	0000001b    0
0000002b explorer.exe
	0000002d    0
	0000002c    0
0000002e (D) C:\Program Files (x86)\CSR\BlueSuite 2.4\BlueFlash.exe
	0000002f    0 <==
System information:
    Wine build: wine-1.6.2
    Platform: i386 (WOW64)
    Host system: Linux
    Host version: 4.4.0-112-generic

teeheee, 2018-02-09:
This is what the spi communication looks like when HC05 is connected.
[image]
And when the HC05 is not connected and the bug not happening.
[image]

lorf, 2018-02-10:
Looks like a bug in driver. Unfortunately I have no HW to test at the moment. If You can assist in debugging, can You please rerun in wine with wine debugging turned on like this:

```env WINEDEBUG=+relay blueflashcmd chipver 2>&1 | grep usbspi```

and post the result.

teeheee, 2018-02-12:
I ran the setup with the following command:

> env WINEDBG=+relay wine BlueFlashCmd.exe chipver 2>&1 | grep usbspi

and it gave this output:

> =>0 0x7e2a6124 spi_xfer+0xfe() in usbspi (0x0033f704)
       1 0x7e2a4897 in usbspi (+0x4896) (0x0033f754)
       2 0x7e2a4fcc spifns_stream_sequence+0xe2() in usbspi (0x0033f794)
0x7e2a6124 spi_xfer+0xfe in usbspi: movzwl	0x0(%eax),%eax
ELF	7e28e000-7e2ba000	Dwarf           usbspi<elf>
  \-PE	7e2a0000-7e2ba000	\               usbspi

teeheee, 2018-03-13:
I found a solution to fix my error. There was some padding problem in SPISEQ_1_4.
I tried to pack the struct but it didn't work so I wrote some ugly code to get it to program. 
Here is the method I changed:

```
DLLEXPORT int spifns_stream_sequence(spifns_stream_t stream, SPISEQ_1_4 *_pSequence, int nCount)
{
    LOG(DEBUG, "(%d, %p, %d)", stream, _pSequence, nCount);

    int nRetval=0;
	    
    unsigned short* pshort = (unsigned short*)_pSequence; //save the address in convinient datatype
    while (nCount--) {
	SPISEQ_1_4 *pSequence = (SPISEQ_1_4 *)pshort;
	SPISEQ_1_4 Sequence; 
	Sequence.nType = pSequence->nType; //first element has no padding
	Sequence.rw.nAddress = pshort[2]; //enum is 32 bit so offset is 4 byte
	Sequence.rw.nLength = pshort[3];  
 	unsigned short** ppshort = (unsigned short**)pshort;
	Sequence.rw.pnData = (unsigned short*)ppshort[2]; // 32 bit address

	if(_pSequence->nType==0 || _pSequence->nType==1) //one type is diffrent?
	    pSequence = &Sequence;

        LOG(DEBUG, "command %d", pSequence->nType);
        switch (pSequence->nType) {
        case SPISEQ_1_4::TYPE_READ:
            if (spifns_sequence_read(pSequence->rw.nAddress,pSequence->rw.nLength,pSequence->rw.pnData)==1)
                nRetval=1;
            break;
        case SPISEQ_1_4::TYPE_WRITE:
            if (spifns_sequence_write(pSequence->rw.nAddress,pSequence->rw.nLength,pSequence->rw.pnData)==1)
                nRetval=1;
            break;
        case SPISEQ_1_4::TYPE_SETVAR:
            if (spifns_sequence_setvar(pSequence->setvar.szName,pSequence->setvar.szValue)==1)
                nRetval=1;
            break;
        default:
            LOG(WARN, "Sequence command not implemented: %d", pSequence->nType);
            g_nError = SPIFNS_ERROR_INVALID_PARAMETER;
            snprintf(g_szErrorString, sizeof(g_szErrorString),
                    "sequence command %d not implemented", pSequence->nType);
            nRetval = 1;
        }
        pshort+=6; // increment by 6 shorts is 12 byte is one package
    }
    return nRetval;
}
```

I do not suggest using this code, but maybe someone who knows the project better can fix the padding problem.

lorf, 2018-03-27:
Thanks for your analysis and code! It looks like BlueSuite 2.4 uses SPI API 1.3, but calls `spifns_stream_sequence()`, which should support that case.

I added appropriate changes to issue-28 branch. If You still have time and HW, can You please test this branch or the precompiled binaries here: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a2 ?