lorf/csr-spi-ftdi #29: FT230X Support

coliniuliano, 2018-03-23:
Hi,

FT230X support appears to be commented out from the source code.  Is there a reason?  With the FT233R chips broken bit bang modes, FT230X seems like a really good replacement.

Thanks

lorf, 2018-03-26:
It was commented out because it was not available in the libftdi version I used at the moment I added it. Seems to be simple to make it a conditional at compile time. I'll look into it.

lorf, 2018-03-27:
Enabled FT230X support in devel branch. Precompiled binaries for testing: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a1

lorf, 2018-03-28:
Updated to mention whole FT-X family of chips, You will need to set `hwspi` pinout (https://github.com/lorf/csr-spi-ftdi/blob/master/README.md#options) to use FT230X. Precompiled binaries: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a4

lorf, 2018-03-30:
I thought it would be easy ;) Will check FT230X docs...

Overflow is fixed in devel branch, thanks!

lorf, 2018-03-31:
There is some changes WRT FT230X in libftdi-1.4, can You please, try the binaries compiled with this version of liibftdi: https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a5

coliniuliano, 2018-04-08:
Hi,

The new version appears to work well, thank you!  I realize now that the reason my chip did not initially identify is because I was using nvscmd.exe from the CSR64xxx tools instead of from BlueModify.  I'm still unable to get the firmware working, but I can confirm that the FT230X support works!

Thanks!

closed 2018-04-08 by coliniuliano