lorf/csr-spi-ftdi #30: Where can you get Bluesuite now?

witnessmenow, 2018-06-07:
Seems like bluesuite downloads has been removed from the CSR website, at least for me anyways.

Is there an alternative place to download it?

lorf, 2018-06-07:
Yeah, they withdrawn all access from my account too. You, probably, need an old good search.

witnessmenow, 2018-06-07:
Thanks for the reply,

I actually still have the files but I have a guide on how to update the HC-05/06 and I was wondering was there an alternative place I could link to.

ridvanaltun, 2018-06-18:
Bluesuite 2.5: https://drive.google.com/file/d/16EHZMVfA7HVyTRGkizt_Gchh8x9HHWx1/view?usp=sharing

Bluesuite 2.6.6: https://drive.google.com/file/d/1ADdvH-hdZSPf3rA8kCM57U-xKHWnCIEp/view?usp=sharing

I hope these links are helpful to anyone.

kennyjudo, 2018-07-03:
Thanks for sharing, by the way, I have a CSR8510 A10 USB dongle, how can I command the dongle? where can I have the command list? I would like to use python for a small project

fantom3l, 2019-02-03:
For those in need, that like me have searched for BlueSuite, i have uploaded a few versions that i found around , even the latest at this point BlueSuite 2.6.11 build 1937
here: https://yadi.sk/d/QAdQ7zuP-X62Xg

el-samiyel, 2019-04-15:
Hi all, does anybody know where I can get bluesuite version 2.1?

Road-Drum, 2019-04-22:
I'm kinda being picky, but is there any known build Bluesuite for Linux? I feel more comfortable on programming this thing on Linux but if only Windows builds are available, I'll just live with it.

and thanks @fantom3l for the latest version available

Alverzilin, 2019-11-17:
Hello friends! I really need version 2.6.2! help me please

RayoSP, 2019-12-01:
http://www.cisdo.cn/post/5.html

ssashton, 2020-10-24:
I have Bluesuit 2.6.6. I have sucsesfully used it to change the name on my CSR8675 modules in the past. I got a new one today but it only says 'Chip not recognised' when I use PSKey or BlueFlash. BlueTest says unable to open SPI.

Does this mean the chip is really not recognised, or can this be locked firmware behaviour?

P1on3R91, 2020-10-27:
Hi to all.
Who know where can download oldest version 2.3, or another lower then 2.4?
It need for BlueCore 2 (BC212015).
Thanks!

Matheus-Garbelini, 2020-10-27:
Anyone have any version of BlueLab software development kit (SDK)?

ssashton, 2020-10-28:
> Hi to all.
> Who know where can download oldest version 2.3, or another lower then 2.4?
> It need for BlueCore 2 (BC212015).
> Thanks!

Fantom3l has some stuff, linked above. https://yadi.sk/d/QAdQ7zuP-X62Xg

ssashton, 2020-10-28:
> I have Bluesuit 2.6.6. I have successfully used it to change the name on my CSR8675 modules in the past, which used the CSR8675 BGA chip.
> 
> I got a new module today that uses the CSR8675 WLCSP chip (smaller IC, fewer pins) but it only says 'chip not recognised' when I use PSKey or BlueFlash. BlueTest says unable to open SPI comunication.
> 
> Does this mean the WLCSP version of the CSR8675 chip is not recognised by BlueSuite 2.6.6, or could this be locked firmware behaviour? I'm not sure how a locked firmware behaves.
> 

I got it figured out. It was simply that this module needed the SPI-PCM pin pulled high to enable SPI mode. Then it was supported by BlueSuite 2.6.6

P1on3R91, 2020-10-28:
> > Hi to all.
> > Who know where can download oldest version 2.3, or another lower then 2.4?
> > It need for BlueCore 2 (BC212015).
> > Thanks!
> 
> Fantom3l has some stuff, linked above. https://yadi.sk/d/QAdQ7zuP-X62Xg
It hasn't  version 2.3, i already saw it ((

CDRXavier, 2021-08-19:
Looks like a copy of the 2010 firmware for RN-42 is not recognized by the PSTool despite BlueFlash worked quite well.

ArtTechGuy, 2022-04-30:
Guys, i hope here is still someone who are interested in this topic
I got the problem with fleshing up the HC 05 with RN 42 firmware
When i downloaded a rn42.xdv and go to PStool i see this picture: 
[image]
Its just an *Error* 
Even if i try to merge dump from HC 05 i see this:
[image]
Why does it happen?

frenchiveruti, 2022-12-08:
I've created an archive.org backup in case the drive files go down.

breakboyandre, 2023-04-03:
@ArtTechGuy same problem here. Did you solve it? Looks like something wrong with bluetooth chip or it's version

ArtTechGuy, 2023-04-03:
> @ArtTechGuy same problem here. Did you solve it? Looks like something wrong with bluetooth chip or it's version

Yes, i solved it, u need to get proper HC05 board with right chip. There is somewhere on intenet the topic of how to identify "right" chip so it can be flashed with rn42 firm. Ill check it later and send u a link, im not in PC rn

breakboyandre, 2023-04-06:
@ArtTechGuy did you find the info about how to identify the right hc-05 module to flash?

breakboyandre, 2023-04-06:
@ArtTechGuy thank you so much

bulentperktas, 2023-09-13:
it seems that it is so hard to find any kind of documentation or tools about Bluecore4-External chips. I found that Bluesuite 2.6.12.3289 version on Qualcomm site but they do not let you use without their customer ID.

And their documentation starts from Qualcomm® BlueCore™ 5 Multimedia External chips, There are not neither BC2, BC3 nor BC4.

Does anybody have this version that 2.6.12?

https://www.qualcomm.com/products/technology/bluetooth/bc57e#Software

Thanks any advise,
Bulent

MDVelkov, 2025-04-17:
> Seems like bluesuite downloads has been removed from the CSR website, at least for me anyways.
> 
> Is there an alternative place to download it?
[
](https://support.avantree.com/hc/en-us/articles/39984893796633-Firmware-Update-for-DG60L)