lorf/csr-spi-ftdi #31: Issue with the FT232RL ("Error detecting chip type...")

evil-m, 2018-06-15:
Hello everybody!
I'm getting an issue with the Blue Flash saying the following message: 
"Error detecting chip type (Unable to start read (invalid control data)).
Select a SPI transport to attemp reconnect."

This is what I have done/find out so far:

- I've already try using the BlueSuite 2.5 and 2.6.2, both end in the same issue;

- I've already try a setup with external psu (5V@2A cellphone charger) and shared GND;

- Did the wiring with and without resistors;

- I'm pretty sure my ftdi chip is a counterfit one;

- Try the Blue Flash application on two different computers (WIN 7 & WIN10);

- Try using 2 bluetooth modules I have (HC-05, HC-06), both with the BC417 143BGC K415AP;

-  The markings on my ftdi chip are: 1119-C GND82401 FT232RL (On the Blue Flash it appears as FT232R A500C697);

- There drivers where installed sucessfully and I didn't found any issue on device manager;

- Double checked the wiring;

- Both bluetooth modules are working fine;

All my tries ended in the same way with the same error message stated earlier in this post.

Both BlueFlashCmd.exe -chipver & BlueFlashCmd.exe -identify have failed as can be seen on the logs below;

Any ideas to solve this?

Thank you guys!

Here are my logs:

**BlueFlashCmd.exe -chipver**

blueflashcmd, version 2.5.0.93 Release
Copyright (C) 2002-2012, Cambridge Silicon Radio Ltd.

17:46:46.929617: all:spi.c:567:spi_init: csr-spi-ftdi 0.5.3-a5, git rev 1f342f2, libftdi 1.4-unknown
17:46:47.304387: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.304387: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.306386: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.306386: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.309384: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.309384: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.313382: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.313382: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.318379: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.318379: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.321377: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.321377: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.324375: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.324375: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.327373: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.327373: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.330371: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.330371: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.333368: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.333368: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.338366: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.338366: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.342364: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.342364: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.345362: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:46:47.345362: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:46:47.348360: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
[... 215 lines cut ...]
17:50:54.064152: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:50:54.064152: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:50:54.067150: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:50:54.068151: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
17:50:54.073147: err:basics.cpp:461:spifns_sequence_read: Control data: 0xff 0xff
17:50:54.074145: err:basics.cpp:483:spifns_sequence_read: Unable to start read (invalid control data)
==============================================================================
ERROR: Error detecting chip type (Unable to start read (invalid control data))
==============================================================================
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3-a5 (git rev 1f342f2)
libftdi version: 1.4-unknown
Time open: 0.23 s
Time in xfer: 0.00 s (0.00% of open time)
Reads: 47 (94 bytes, 2.00 bytes avg read size)
Writes: 47 (141 bytes, 3.00 bytes avg write size)
Xfer data rate: 1.#R KB/s (235 bytes in 0.00 s)
IOPS: 1.#R IO/s (94 IOs in 0.00 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1.#R xfers/s (1.#R short reads/s,
            116 xfers/20 short reads in 0.00 s,
            1.00 xfers/IO, 69.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Failed

EdKok, 2018-06-16:
Sorry, can't help. Having exact same problem.
Using Win-10/64-bit.

*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3-a5 (git rev 1f342f2)
libftdi version: 1.4-unknown
Time open: 0.23 s
Time in xfer: 0.00 s (0.00% of open time)
Reads: 52 (104 bytes, 2.00 bytes avg read size)
Writes: 52 (156 bytes, 3.00 bytes avg write size)
Xfer data rate: 1.#R KB/s (260 bytes in 0.00 s)
IOPS: 1.#R IO/s (104 IOs in 0.00 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1.#R xfers/s (1.#R short reads/s,
            133 xfers/27 short reads in 0.00 s,
            1.00 xfers/IO, 67.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Failed

evil-m, 2018-06-16:
I only tried Blue Suite tools on x64 based windows, later today I'll test it on a x86 based windows to see if there are any changes atl all.
Maybe this fail is related to those chips with the bitbang issue as stated [here](http://blog.bitheap.net/2012/03/ft232r-bitbang-mode-is-broken.html).
I guess my chip is revision C (1119-C marked on it), but I can't say for sure. I'll take a look upon this later.

If you manage to solve your issue, please, let me know. If I found any working solution I'll be glad to share it here.

Thank you!

lorf, 2018-06-17:
Hi! Such a failure means complete communication failure between the host and CSR chip (cannot find the signature in chip's response) and can be caused by anything from driver bug to faulty FTDI chip to wiring faults to faulty CSR chip, etc. So I cannot really help here.

However I think it's not caused by x64 based Windows, and I believe counterfeit FT232RL's are not affected by FTDI's hardware bug.

evil-m, 2018-06-18:
If I plug the FT232RL on the usb without any CSR bluetooth chip connected to it I should expect a message like the "Error detecting chip type..." or it shows anything different than this?

Thank you!

ridvanaltun, 2018-06-18:
@evil-m , "Error detecting chip type..." showed up. I have the same problem too, i'm using windows 10 and i will test it in windows 7 soon.

evil-m, 2018-06-18:
Just tried on x86 based windows too, got the same result. 
I double checked the wiring, everything seems to be okay. I'll try the schematic for LPT port to see if I manage to get Blue Flash working, this way I'll be able to exclude the ft232rl from the equation and verify if the connection failure gets away, if not, I guess my bluetooth modules are the problem.

evil-m, 2018-06-19:
Hey guys, I just found that the release [0.5.1](https://github.com/lorf/csr-spi-ftdi/releases/download/0.5.1/csr-spi-ftdi-0.5.1.zip) allows me to dup the firmware with BlueFlash and backup the settings with Pstool, although, when I try to flash (download) my backup firmware back to the module, BlueFlash says: "Device 0: Failed to erase sector 0 after 4 attempts. Abandoned"
I've tested it on a Win 7 x64 with Blue Suite 2.5 & 2.6.2. Any hints?

Thank you!

ridvanaltun, 2018-06-19:
It's alive! I have tested it on Windows 10 x64 with the release 0.5.1 and Bluesuit 2.6.6. I wrote the RN42 firmware on HC-05. HC-05 works as keyboard now. Thanks for the information.

@evil-m , @EdKok , @lorf

evil-m, 2018-06-19:
@ridvanaltun

Do you have a download link or could you upload the bluesuite 2.6.6 somewhere?

About the releases I've tested with Blue Suite 2.5 & 2.6.2:
- 0.5.3:  None of them worked, showing issues, as described in this post;
- 0.5.2:  Pstool gives reading erros;
- 0.5.1:  Worked for dumping the firmware and backing up the settings but I got the "Device 0: Failed to erase sector 0 after 4 attempts. Abandoned" message while trying to download a firmware to the module, also the blueflashcmd -verify & blueflashcmd -identify shows that everything is ok, but sometimes -identify gives me the following error:

I only tested it on my HC-05, I was unable to test it on the HC-06 because one pad got broken when the module fell on the floor yesterday. Does any one have a schematic for this boards? (Maybe I could solder a enameled wire if I found the broken pad's tracks somewhere on the board.) The pad that got broken is the CLK and I cannot see where the track goes because it cames from the middle of the fiberglass and passes underneath the flash chip.

evil-m, 2018-06-19:
I only tested it on my HC-05, I was unable to test it on the HC-06 because one pad got broken when the module fell on the floor yesterday. Does any one have a schematic for this boards? (Maybe I could solder a enameled wire if I found the broken pad's tracks somewhere on the board.) The pad that got broken is the CLK and I cannot see where the track goes because it cames from the middle of the fiberglass and passes underneath the flash chip.

ridvanaltun, 2018-06-19:
You can find Bluesuit 2.6.6 [here](https://github.com/lorf/csr-spi-ftdi/issues/30).

I hope it works. 
Best wishes.

evil-m, 2018-06-19:
@ridvanaltun Thank you! Soon as I manage to test it I'll post the results here!

EdKok, 2018-07-07:
@ridvanaltun Thanks for the reply.
Today again tried anything I can think of but nothing works. Seems I have the same setup as you have (software wise) but keep getting the same errors. Used two different HC-05's and two different FT232RL's. Just for this little project bought new ones. Could you perhaps post links where you have your hardware from? And how did you make the connections between the devices? I'm using 100ohm resistors, without any patching leads to keep connections short. Also tried lowering the speed.

ridvanaltun, 2018-07-07:
Hello @EdKok , mine bt module is very old, it is 3 years old so the hardware supplier doesnt exist any more.  

I used 220ohm resistors, there is nothing for special. 

Here is my tiny hardware setup:

[image]
[image]
[image]

Best wishes.

BoKKeR, 2018-09-03:
Release 0.5.1 fixed the issue for me too.

BoKKeR, 2018-09-03:
Actually using 2.6.6 + 0.5.1 managed to get the chip recognized but when I try to dump the FW I get "No Flash detected" and when I run the identity command I get "Unable to calculate addressing mode of EEPROM". Anything I should do? maybe erase the flash?

lorf, 2018-11-27:
Hi! There was a bug with wrong default pinout in development versions of the driver (0.5.3-a*). It was fixed in [0.5.3-a9](https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a9). See #39.

evil-m, 2019-01-25:
I guess that the "Error detecting chip type..." problem were related to the issue #39, I don't get this message with any releases other than version 0.5.3. But I'm not able to flash any firmware to the module. My modules don't have breakout board the ones I have are like this [one](https://www.rhydolabz.com/images/WRL2106.jpg). There are any pin that I have to set HIGH or LOW, or make some kind of pull UP/DOWN in order to enable flash writing? 
At the momment I only have connected GND, 3.3V, CLK, MISO, MOSI, CSB.

Thanks!

GluTbl, 2019-08-17:
I

> Hi! There was a bug with wrong default pinout in development versions of the driver (0.5.3-a*). It was fixed in [0.5.3-a9](https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a9). See #39.

I don't know how it work..... i think i use think "0.5.3-a9" and worked

danielhc144, 2019-12-09:
Yo logré evitar este problema utilizando la versión 0.5.2 https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.2

marcoA152, 2021-01-09:
Hi to everyone, i'm intersted to this project but i have this issue:
I use latest release (csr-spi-ftdi-0.5.3-a9) 
bluesuite ( ver 2.6.6 and 2_6_9_1584) 
asus laptop with win10pro x64, 
zadig drivers libusbK 3.0.7.0
on the FT232RL chip i can read 1922-C
also tried this from the documentation https://github.com/lorf/csr-spi-ftdi/blob/master/misc/spi-set-usb-transport.reg
i've checked too much times wiring, carefully soldered 4 thin wires to hc05 pins
so i receive this error with and without hc05 connected 

**for you is a wiring issue or is a misconfig of the FT232RL ??**

thank you

BlueFlashCmd.exe -chipver
blueflashcmd, version 2.6.6.1311 Release
Copyright (c) 2002-2017 Qualcomm Technologies International, Ltd.
All Rights Reserved.
Qualcomm Technologies International, Ltd. Confidential and Proprietary.

22:15:21.026744: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.1, git rev b60665a
22:15:21.154376: err:spi.c:166:spi_ftdi_xfer: ***************************************************
22:15:21.154604: err:spi.c:167:spi_ftdi_xfer: Lost 1 of 1 bytes of data in transit
22:15:21.154604: err:spi.c:168:spi_ftdi_xfer: Probably a counterfeit FT232RL in USB3.0 socket.
22:15:21.154604: err:spi.c:169:spi_ftdi_xfer: Try to plug programmer into USB 2.0 socket.
22:15:21.154604: err:spi.c:170:spi_ftdi_xfer: ***************************************************
22:15:21.205357: err:spi.c:166:spi_ftdi_xfer: ***************************************************
22:15:21.206358: err:spi.c:167:spi_ftdi_xfer: Lost 86 of 86 bytes of data in transit
22:15:21.206358: err:spi.c:168:spi_ftdi_xfer: Probably a counterfeit FT232RL in USB3.0 socket.
22:15:21.206358: err:spi.c:169:spi_ftdi_xfer: Try to plug programmer into USB 2.0 socket.
22:15:21.207360: err:spi.c:170:spi_ftdi_xfer: ***************************************************
22:15:21.207781: err:basics.cpp:479:spifns_sequence_read: Unable to start read (getting control data)
22:15:26.214528: err:spi.c:118:spi_ftdi_xfer: FTDI: write data failed: [-1] usb bulk write failed
22:15:26.214528: err:basics.cpp:479:spifns_sequence_read: Unable to start read (getting control data)
==============================================================================
ERROR: Error detecting chip type (Unable to start read (getting control data))
==============================================================================
22:15:31.230674: err:spi.c:118:spi_ftdi_xfer: FTDI: write data failed: [-1] usb bulk write failed
Failed

mazWaz, 2022-02-19:
> Hi to everyone, i'm intersted to this project but i have this issue: I use latest release (csr-spi-ftdi-0.5.3-a9) bluesuite ( ver 2.6.6 and 2_6_9_1584) asus laptop with win10pro x64, zadig drivers [...]

I have exactly same error, any solution?

Kapoa, 2022-08-02:
I am also getting the same issue reading from an HC 05 chip on windows 11. tried csr-spi-ftdi 0.5.1, 0.5.3, 0.5.3-a9 with BLueSuit v2.6.6.1311 and v2.4
and changed the bitrate. not had any luck.

`C:\Program Files (x86)\CSR\BlueSuite 2.6.6>BlueFlashCmd.exe -chipver
blueflashcmd, version 2.6.6.1311 Release
Copyright (c) 2002-2017 Qualcomm Technologies International, Ltd.
All Rights Reserved.
Qualcomm Technologies International, Ltd. Confidential and Proprietary.

12:47:13.755527: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.1, git rev b60665a
12:47:13.840386: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.842963: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.844964: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.848969: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.851963: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.853963: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.856398: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.859401: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.862972: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.864853: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.871960: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.874962: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.877964: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.880973: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.882963: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.884972: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.890952: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.892953: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.895976: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.897956: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.899964: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.903953: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.910531: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.912553: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.915537: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.917959: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.919783: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.923528: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.926531: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.928568: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.930541: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.936291: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.939539: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.943539: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.947529: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.949529: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.952529: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.955567: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.958569: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.960581: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.964544: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.967547: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.970557: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.973530: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.976557: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.981531: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.984536: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
12:47:13.986579: err:basics.cpp:479:spifns_sequence_read: Unable to start read (invalid control data)
[... cut ...]
FTDI stats: 1.#R xfers/s (103023958578841360000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.00 short reads/s,
            156 xfers/26 short reads in 0.00 s,
            1.00 xfers/IO, 70.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Failed`

also of note is in BlueFlash my FT232RL programer comes up as __FT232R 00000000__

Is there a way to rule out a hardware fault in the FTDI chip?

Jog-fall, 2024-10-06:
Hello, I am very new to this stuff but i am getting the error :-
[image]
I am trying to follow this turtorial:-
https://www.youtube.com/watch?v=y8PcNbAA6AQ
Please help asap

ceut2k4, 2024-12-07:
Hello everyone,
I was having this issue with a red PCB FT232RL module I have bought some months ago.
Finally mine was a fake one, that's why I have had all these errors.
I have made a little video here where I have put all in the text description.
https://www.youtube.com/watch?v=4XwhcjlG4hg

Since I swapped the fake FT232RL with the genuine FT232RNL: all is working straightforward and flawlessly.

=>Check your serial number, it is the only thing to do.