lorf/csr-spi-ftdi #37: v 0.5.3-a5 and Wine 3.0-1ubuntu1 problem

T-2, 2018-10-08:
Having seg fault errors with the newest version of your driver. Older 0.5.2 works just fine. Have not tried the versions in between. Wine 32 bit v 3.0-1ubuntu1 on Linux Mint 19 Mate, CSR BlueSuite v2.6.4.

History:
Had been using the 0.5.2 driver on several versions of Linux Mint over a few years. Always used Bluesuite v2.6.4 and Wine v 1.6.? Once I got it setup it always worked. Just used PSTool, BlueFlashCmd, and pscli from CSR.

I then decided to upgrade from Mint 18.3 to 19 to get all the cool new bells and whistles. The upgrade went fine except for wine. The default install for wine on 19 wants to install the 64 bit version of wine. BlueSuite does not like that version it turns out. After several tries I found a combination that allowed the CSR software to install under wine.
FYI the wine install:
 sudo dpkg --add-architecture i386 && apt update
 sudo apt install wine-stable wine32 wine64 libwine libwine:i386 fonts-wine
 WINEARCH=win32 WINEPREFIX=~/.wine wine wineboot  (creates the .wine folder for 32 bit apps)

After that when installing Bluesuite it did complain about something missing from winbind so I installed winbind from the package manager and reran the Bluesuite install and got no more error.

I then did your procedure to install your usb driver v 0.5.3-a5. When I ran both PSTool or pscli they recognize the FT232R chip but would seg fault almost immediately. BlueFlashCmd did not seg fault but did not seem to recognize the chip.
Then just for giggles I tried the driver version of yours I have always been using and it works just fine.

I hope this helps others reduce the time it takes to getting this to work.
Thanks for your work on this driver. I can still be Windows free! - for the most part.

lorf, 2018-11-25:
Yes, 0.5.3-a# are development versions, and may not work as expected. 0.5.2 is the latest "stable" version.

T-2, 2019-05-08:
Same problem with the release version 0.5.3. Also tried with wine v4 and still no go. I have yet to see if I slow the speed down if it will behave. Not sure as it fails with PSTools, pscli, and BlueFlashCmd. Where version 0.5.2 works fine.

closed 2019-05-08 by T-2