lorf/csr-spi-ftdi #38: FTDI reading 0xdeaf always

alberto-fc, 2018-10-17:
Hello,

I hit a wall with a CSR8675, every read returns:

```
[2164504327]read ff9a  Failed
10:00:26.478968: dump:basics.cpp:470:spifns_sequence_read: read16(addr=0xff9a, len16=1) (size=2, crc32=0xf4e7e8bb):
  0000  af de                                            ..
[2164504376]read ff9a deaf
10:00:26.525843: dump:basics.cpp:470:spifns_sequence_read: read16(addr=0xff9a, len16=1) (size=2, crc32=0xf4e7e8bb):
  0000  af de                                            ..
[2164504425]read ff9a deaf
```

There is also an issue with clocks (clocks on FTDI is always x4 the speed regardless of the parameter -20kHz = 80kHz-) and last versions (0.5.2 and 0.5.3 doesnt work with FT232RL, it seems the pin doesn't match the <0.5.1).

In any case, the problem checked with the oscilloscope is the reading of 0xdeaf.

Any idea why this is happening ?

Thank you.

lorf, 2018-11-23:
The pinout was fixed in [0.5.3-a9](https://github.com/lorf/csr-spi-ftdi/releases/tag/0.5.3-a9). No idea about 0xdeaf, sorry.

alberto-fc, 2018-11-24:
Hi there,

I've found quite interesting things about this.

First, plain, and simple: 0xdeaf it's a "non-public-documented" response for CSR8675 and CSR8670 (probably more ICs from CSR used the same response-command) when the device is "protected". It was confusing (and extremely irritating) because it was my first contact with this and I could not determine where the error or bug was coming from (the change of versions also changed the answer because the pinout 'bug' and added it even more confusing to the pile).

CSR provides a tool (it comes with BlueSuite) to "unlock" the device (actually, there are two ways to do it, nvscmd and spiunlock, you need a "key" or you need to full-erase the device).

What this means ? You lost APT-X fuses and "all the functionality" programmed in the module. It is pain, but you can use the chip later for your "purposes".

Also, if you want I can open another issue about the clocking (it re-checked everything, the SPI clock isn't working as stated in the commands, it's **always four times** the clock speed regardless the parameter).

A.

lorf, 2018-11-25:
Thanks for the info! I'll need to add it to the README.

Yes, please open another issue about the clocking.

Pieter60, 2020-09-09:
Same problem here.
Bluetooth device CSR8675 with aptx-HD.

Used the LPT-interface.
Command "spiunlock.exe unlock dump.txt -lpt 1"
Give some more info.

Content of dump.txt:
12345678901234567890123456789012

The csr-spi-ftdi interface does not function with a locked device.

Can someone confirm if a locked device can be eared by means of command:
"spiunlock.exe erease -lpt 1"

alberto-fc, 2020-09-09:
> Same problem here.
> Bluetooth device CSR8675 with aptx-HD.
> 
> Used the LPT-interface.
> Command "spiunlock.exe unlock dump.txt -lpt 1"
> Give some more info.
> 
> Content of dump.txt:
> 12345678901234567890123456789012
> 
> The csr-spi-ftdi interface does not function with a locked device.
> 
> Can someone confirm if a locked device can be erased by means of command:
> "spiunlock.exe erase -lpt 1"

> CSR provides a tool (it comes with BlueSuite) to "unlock" the device (actually, there are two ways to do it, nvscmd and spiunlock but you need a "the programmed key" or you have to full-erase the device).

I can confirm you that at least in my devices, the erase allowed me to reprogram the modules later (and stopped the 0xdeaf response).

Pieter60, 2020-09-09:
Thanks for the info Alberto.

I leave this locked chip in tact and later on I will buy a cheap CSR8675 module.
Try to program it with aptx-HD enabled FW.

alberto-fc, 2020-09-09:
> Thanks for the info Alberto.
> 
> I leave this locked chip in tact and later on I will buy a cheap CSR8675 module.
> Try to program it with aptx-HD enabled FW.

In the end i finished erasing the APT CSR chipset I have, because I could find some tools to "reprogram" the fuses and give 'em the APTX funcionality again. Not easy task I coudl tell (CSR-SDK/Bluelab csr but you need the "fuses" for each chip, and thats not so easy without weird chinese forums).

Anyways, my modules cost around 7-10€ each, so I was ready to sacrife 'em to the good, but I was glad that I'd able to recover the APT funcionality on some on them.

My main and final application for all of this, I was to use the module (CSR) plugged to the computer as HID-USB device and allow me to use the play/resume, vol+/vol- of the headset. But I wasn't ready to sacrify the APTX for that...

Pieter60, 2020-09-09:
My module has aptx-HD enabled.
I want/have to change the default pin-code.
The problem is that the module is completely deaf.
SPI locked, USB not functioning, UART not functioning, BT serial port not functioning.

ShtokyD, 2024-07-11:
> > I think you can send me a personal message from github (I can't promise anything :D, but maybe I can serve you some hints).
> 
> Thank you! I do not know how to personal message. I'm new to Github (and also coding!).
> 
> Maybe you can drop me an email at ssashton AT gmail dotcom and then we can talk? Today I managed to get PSTool to talk with my CSR8675.

Hi, how did you manage to bypass the locked SPI?

ssashton, 2024-07-11:
Sorry it was a long time ago I don't remeber what I did!

ShtokyD, 2024-07-11:
> Sorry it was a long time ago I don't remeber what I did!
> […](#)

Oh noooo haha, that would help me a lot, because I'm also stuck with a locked SPI access to the CSR8675 chip....