lorf/csr-spi-ftdi #39: Wrong wiring table in readme?

DeepSOIC, 2018-11-16:
Hi!
I am trying to hack my JBL Flip 4, which has CSR8675 inside.

I have soldered up an ftdi thing according to this table I found in readme:

| Signal | FT232RL pin | FTDI pin name | FTDI GPIO bit | CSR pin  |
|--------|-------------|---------------|---------------|----------|
| CS#    | 2           | DTR#          | D4            | SPI_CS#  |
| CLK    | 3           | RTS#          | D2            | SPI_CLK  |
| MOSI   | 6           | RI#           | D7            | SPI_MOSI |
| MISO   | 9           | DSR#          | D5            | SPI_MISO |

PSTool says "error detecting chip type".
After a few checks and such, I started probing with scope to see what's going on. And found out, that neither of the FTDI pins I soldered wires to give off any signals. Instead, I'm seeing what looks like CLK on Tx, and what looks like MOSI on Tx, and what looks like CS on CTS. Now assuming I got these pins sorted out, which one should I use as MISO?

DeepSOIC, 2018-11-16:
Working!

The wiring that works for me:

| Signal | FTDI pin name |
|--------|---------------|
| CS#    | CTS#          |
| CLK    | Tx            |
| MOSI   | Rx            |
| MISO   | RTS#          |

IDK if it's relevant, but I'm on Win10 64-bit, PSTool v.2.6.6.1311
I downloaded csr-spi-ftdi-0.5.3-a5.zip, associated with commit 1f342f246223669d6db8138367633f1f0f9b67f5

DeepSOIC, 2018-11-17:
Thanks for quick fix, and for the tool. I'd test it, but I somehow busted my chip in a matter of an hour =( 
If curious, see https://hackaday.io/project/162337-reverse-engineering-jbl-flip-4

DeepSOIC, 2018-11-19:
The chip turned out to be fine. I tested the new release (csr-spi-ftdi-0.5.3-a9) with pinout in readme, and it works. Thanks.

lorf, 2018-11-20:
Thank You for testing!

LuthfiZXC, 2021-05-18:
> Working!
> 
> The wiring that works for me:
> 
> Signal	FTDI pin name
> CS#	CTS#
> CLK	Tx
> MOSI	Rx
> MISO	RTS#
> IDK if it's relevant, but I'm on Win10 64-bit, PSTool v.2.6.6.1311
> I downloaded csr-spi-ftdi-0.5.3-a5.zip, associated with commit [1f342f2](https://github.com/lorf/csr-spi-ftdi/commit/1f342f246223669d6db8138367633f1f0f9b67f5)
> 
> Using FT232RL (IDK if genuine or not) in SSOP28 package, on own board, at 3.3V VCCIO (powered from Flip4's board)

I used QCC3008 module and it works for me, thank you very much

commit f586fe15 (2018-11-17): Fix default pinout

An error was introduced in 831e117 - "HWSPI" pinout was used by default
instead of the default one described in README. Fixes #39