lorf/csr-spi-ftdi #41: RN-42 Firmware Truncation

rupin, 2019-03-13:
When I chose the RN-42 Firmware available here https://drive.google.com/file/d/0B2oHzSfiLL8PdU1tWmhvS19EWG8/view

Blueflash Reports that the image size is bigger than flash size, and the firmware gets truncated. 

"Image size is greater than Flash Size. Image will be truncated"

The upload happens, but PS Tool does not list the data from the persistent storage. 
[image]

If anyone has seen this issue happen, please advise how to solve it.

lorf, 2019-03-13:
See issue #25

lorf, 2019-03-13:
BTW, while counterfeit modules described in #25 are based on a relabeled "BC3 MM" chip (codenamed "KAL") and have 1 MB flash, your module is based on a "BC3 Audio Flash" chip (codenamed "PDW"), which is somewhat newer than KAL, but looks like it has less flash memory. It would be great if You can provide more info about your module, such as pics, output of "e2cmd.exe info" and "BlueFlashCmd.exe identify", when run in a terminal.

lprot, 2019-08-23:
Got the same chip:
c:\Program Files (x86)\QTIL\BlueSuite 2.6.11>e2cmd info
e2cmd, version 2.6.11.1937 Release
Copyright (c) 2007-2018 Qualcomm Technologies International, Ltd.
All Rights Reserved.
Qualcomm Technologies International, Ltd. Confidential and Proprietary.

13:26:32.128657: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
Chip ID - 0x07e2
Chip Name - BC3 Audio Flash (pdw)
Unable to calculate addressing mode of EEPROM
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 8.95 s
Time in xfer: 8.64 s (96.46% of open time)
Reads: 233 (46432 bytes, 199.28 bytes avg read size)
Writes: 332 (47108 bytes, 141.89 bytes avg write size)
Xfer data rate: 10.57 KB/s (93540 bytes in 8.64 s)
IOPS: 65.00 IO/s (565 IOs in 8.64 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 955.00 xfers/s (0.00 short reads/s,
            8255 xfers/1 short reads in 8.64 s,
            14.00 xfers/IO, 363.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************

c:\Program Files (x86)\QTIL\BlueSuite 2.6.11> BlueFlashCmd.exe -chipver
blueflashcmd, version 2.6.11.1937 Release
Copyright (c) 2002-2018 Qualcomm Technologies International, Ltd.
All Rights Reserved.
Qualcomm Technologies International, Ltd. Confidential and Proprietary.

17:03:17.089485: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
GBL_CHIP_VERSION = 0x07e2
Result:
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 0.03 s
Time in xfer: 0.00 s (23.53% of open time)
Reads: 4 (8 bytes, 2.00 bytes avg read size)
Writes: 2 (6 bytes, 3.00 bytes avg write size)
Xfer data rate: 1.71 KB/s (14 bytes in 0.00 s)
IOPS: 750.00 IO/s (6 IOs in 0.00 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1500.00 xfers/s (0.00 short reads/s,
            12 xfers/0 short reads in 0.00 s,
            2.00 xfers/IO, 40.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 1000 kHz, slowdowns: 0
****************************************************************************
Success

c:\Program Files (x86)\QTIL\BlueSuite 2.6.11> BlueFlashCmd.exe -identify
blueflashcmd, version 2.6.11.1937 Release
Copyright (c) 2002-2018 Qualcomm Technologies International, Ltd.
All Rights Reserved.
Qualcomm Technologies International, Ltd. Confidential and Proprietary.

17:04:24.714259: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
Resetting XAP
Identifying XAP
Flash identity: size = 96 sectors (6 Mbit), man_id = 0x0000, dev_id = 0x0000
Firmware ID (loader)="elv_8unified_fl_bt2.1_23g_0903311011_native_encr56 2009-03-31"
Firmware ID (stack)="pdw_6compact_fs_fl_bt2.1_23g_0903311011_encr56_nodfu 2009-03-31"
Result: Usable flash size: 96 sectors, 6 megabit.
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 14.39 s
Time in xfer: 13.93 s (96.79% of open time)
Reads: 408 (75518 bytes, 185.09 bytes avg read size)
Writes: 569 (75074 bytes, 131.94 bytes avg write size)
Xfer data rate: 10.55 KB/s (150592 bytes in 13.93 s)
IOPS: 70.00 IO/s (977 IOs in 13.93 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 956.00 xfers/s (0.00 short reads/s,
            13332 xfers/8 short reads in 13.93 s,
            13.00 xfers/IO, 361.00 bytes/xfer)
SPI max clock: 1000 kHz, min clock: 666 kHz, slowdowns: 1
****************************************************************************
Success