lorf/csr-spi-ftdi #45: Counterfeit problem with latest driver and the DSP

MendesLL, 2019-05-20:
Hi guys, I'm here to say that I got a counterfeit ftdi chip and it has issues with drives after 0.5.1 and this is solvable by just using an older driver, probably because of the bit bang issue that counterfeits have. But, it doesn't work with the UFE(universal front end), that is used to change the DSP parameters(equalizer, CVC, noise cancelling, etc...), I've saw a guy complaining about some functions that are on the UFE driver and that are missing on the FTDI driver.

Was this(DSP comm) corrected on the latter version?
Is there anyway to patch 0.5.3 for counterfeits?

I'm using a counterfeit FTDI232RL and CSR8635 breakout boards.

sidlauskaslukas, 2019-05-22:
@MendesLL make sure you stream the music or have an active call when using UFE

MendesLL, 2019-05-22:
@sidlauskaslukas I did, can you say if you had success using Iorf's drivers with FTDIs and the UFE?
Cause I'm not entirely sure of where the problem could be, and as far as I know, the driver is for bluesuite apps, but I tested it and did well with the Headset Config Tool (a more user friendly version of PStools)

sidlauskaslukas, 2019-05-23:
@MendesLL UFE works for me even with the latest version 5.3.0. How did you wired up CSR8635 to FTDI232RL? Do you get any error message from UFE?

MendesLL, 2019-05-23:
UFE can't connect/find the kalimba DSP, since it works for you, it should be that I'm using an outdated driver cause of my counterfeit. I'll try to set up the latest driver, or to get an genuine FTDI chip

MendesLL, 2019-05-23:
I've made 0.5.3 works with bluesuite and headset config tool, UFE still not connecting, it is strange, no error showing up
Seems that spi is extremely sensible, I'll try some wiring schems, but blueflash still can't dump my chip, says "no flash detected", besides pstools can rename it without bricking, I should probably get some old PC with an LPT port, or a genuine FTDI

sidlauskaslukas, 2019-05-30:
@MendesLL it worked for me even with counterfeit FTDI. Even with the genuine FTDI sometimes the UFE fails to connect, but I found that OS reboot fixes the problem. I also found that powering the BT from external 3.3V supply helps to resolve similiar issues. I was unable to flash new firmware until I powered it from external power supply..

MendesLL, 2019-07-21:
@sidlauskaslukas 
I've been out for days, or months, in the meantime I've abandoned this, catch up again, burned an amp+bluetooth board, got accepted at university.
Would you pls give a link to some post or anything on how to make UFE work?
Or if you still remember describe exactly how UFE worked for you. Using VM? Which SO? How did you wire the module/ftdi (how long/short are the wires, series protection resistor?) Whas you using it on a breadboard? 
If possible(you probably dont have it) would be nice to have a picture of the setup.

closed 2021-12-02 by MendesLL

sidlauskaslukas, 2021-12-08:
@MendesLL did you managed to make UFE work?

MendesLL, 2021-12-08:
@sidlauskaslukas No, I just closed the issue due to inactivity.

Well, bad practice to not close without reporting the reason, but I never managed to make UFE work using the couterfeit FTDI chips. FT2232 or FT232H is the way to go