lorf/csr-spi-ftdi #49: Chip select and SPI clock not working correctly

kcl93, 2020-04-14:
Hello,

I got an issue with communicating with a CSR8645 using a FTDI232R (A94FFLDD) and BlueSuite 2.5.8. I can't get any proper communication and the following error is always returned:

"20:49:41.151676: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)"

Because of this I decided to measure the SPI interface using an oscilloscope which resulted in the following signals:

[image]

The chip select is low most of the time and only high for a short pulse in the beginning of an SPI frame. Also the SPI frequency is always ~77kHz independed from which value I set via command line.
Do you know what might cause this issue and how I can solve this?
Thank you very much in advance!

Kind regards,

kcl_93

kcl93, 2020-04-16:
I have tested it with another FTDI chip that I just bought as well as via a linux VM. All results in the same behaviour.

kcl93, 2020-04-16:
I have found a FT2232HL which should definitively be a genuine part and it shows nearly exactly the same behaviour as above. Just with fixed 25kHz output clock instead of 77kHz,

darkautism, 2020-04-17:
https://www.eevblog.com/forum/microcontrollers/ftdi-gate-2-0/msg859741/#msg859741
Does it related to this?

kcl93, 2020-04-17:
> https://www.eevblog.com/forum/microcontrollers/ftdi-gate-2-0/msg859741/#msg859741
> Does it related to this?

Sadly not so much. The signals I get look fine but the number of clock cycles is odd. I am not sure whether some at the beginning or end are missing or whether more clocks are output than should be.
At the moment I am trying to setup an Eclipse project to compile this code myself as the hints in the readme do not seem to work anymore (or I am not competent enough to get them to work ;) )

jeff-dagenais, 2020-09-13:
I also measure the clock frequency at 80KHz, even if the output says 20kHz

```
"C:\Program Files (x86)\CSR\BlueSuite 2.5.0\BlueFlashCmd.exe" -trans "SPIMAXCLOCK=20" -identify
blueflashcmd, version 2.5.0.93 Release
Copyright (C) 2002-2012, Cambridge Silicon Radio Ltd.

16:56:03.855674: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
16:56:04.386665: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.402157: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.417761: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.433389: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.433389: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.496811: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.543453: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.558411: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
16:56:04.574012: err:basics.cpp:481:spifns_sequence_read: Unable to start read (invalid control data)
==============================================================================
ERROR: Error detecting chip type (Unable to start read (invalid control data))
==============================================================================
*** FTDI Statistics ********************************************************
csr-spi-ftdi version: 0.5.3 (git rev 80b2ad0)
Time open: 0.54 s
Time in xfer: 0.00 s (0.00% of open time)
Reads: 9 (18 bytes, 2.00 bytes avg read size)
Writes: 9 (27 bytes, 3.00 bytes avg write size)
Xfer data rate: 1.#R KB/s (45 bytes in 0.00 s)
IOPS: 1.#R IO/s (18 IOs in 0.00 s)
FTDI chip: FT232R (3), buffer size: 384 bytes
FTDI stats: 1.#R xfers/s (1.#R short reads/s,
            26 xfers/6 short reads in 0.00 s,
            1.00 xfers/IO, 59.00 bytes/xfer)
SPI max clock: 20 kHz, min clock: 20 kHz, slowdowns: 0
****************************************************************************
Failed
```

kcl93, 2020-09-15:
> Any updates?

Nothing from my side. I sadly never got it working...