lorf/csr-spi-ftdi #51: ADK does not work with driver

DSchndr, 2020-05-08:
The new ADK Configuration Tool (for qcc300x) seems to load the driver but then refuses to work with it.
Could be that Qualcomm added something to the driver that is missing here.

DSchndr, 2020-05-08:
The program crashed if you force the name to be "USB SPI" instead of "FT232R xxxxxxxx"
Other utilities work just fine with the changed name.

Maybe there is a check somewhere?

SIDHUNP, 2020-11-19:
hello. i have exact same issue. is there any way to fix it? or did you finally order CSR USB SPI programmer?

DSchndr, 2020-11-20:
With the new adk, solder a usb cable to the data pins of the module and connect 5v+ to vin? pin.
Adk config tool works apparently only with usb and not with a spi programmer.
For spi programmer i made one with a stm32 blue pill which works flawlessly.

SIDHUNP, 2020-11-30:
I will try it thanks.
I tried once with USB, but my board had 3 USB pins namely USBP, USBN, USB3V3. And it had not worked.
I will leave out 3V3 pin and try with data and 5V for VCC. (I am guessing common ground for USB cable and PCB)
PS: For me UFE and PSTool worked no problem with FT232. (Bluesuite 2.6.8)

SIDHUNP, 2020-12-02:
did not work for me yet. as per datasheet, device should have HID(USB) enabled. It is not. and i need ADK config tool for that.
Not sure of drivers too.

DSchndr, 2020-12-03:
I think that i saw that hid can be enabled somewhere in the pskey tool
Do you get a "Qualcomm qcc300x ..." device when you plug it in?
adk is here: https://www.dropbox.com/sh/o4lyrjb1kq73g4j/AADhk5-PCQSe3aND0PfMa5jga?dl=0

SIDHUNP, 2020-12-03:
I didn't get any notification for USB device connection.
I have ADK. I meant that to enable HID I need ADK connected, for which I need HID enabled(deadlock ;).
Okay I will check in PSTool, but yesterday I searched in vain in the Key list.
Thanks!!

DSchndr, 2020-12-03:
Which chip do you have?
I could send a dump of my qcc3003
Iirc on older versions all values were saved in pskeys (aka spi flash)

SIDHUNP, 2020-12-03:
I have QCC3008. if you remember the key name it would be great. i searched for HID and Profile, but did not get a hit.

SIDHUNP, 2020-12-03:
you can send the dump, maybe i will get lucky when i do file compare with mine.
thanks!

LuthfiZXC, 2021-07-18:
> Which chip do you have?
> I could send a dump of my qcc3003
> Iirc on older versions all values were saved in pskeys (aka spi flash)

Can you send your QCC3008 psr dump , please?