lorf/csr-spi-ftdi #54: ("Error detecting chip type...")

HoHy05, 2020-10-26:
Like the title, i keep getting the  ("Error detecting chip type...") message everytimes i tried to open BlueFlash, i have tried BlueFlash from 2.4 to 2.5 - 2.6.2 - 2.6.6 - 2.6.11 and crs spi ftdi from 0.5.0 to 0.5.1 - 0.5.2 - 0.5.3 a9 - 0.5.3 and the error keep popping up, i have even double check the wiring from the pinout, wiring without resistor, tried external power input from battery for the HC-05. I have tested with 2 FT232RL breakout board, both HC-05 and HC-06 but didnt have any chance to get off the error.

[image]

ttm56p, 2021-01-30:
@HoHy05 have you managed to solve it? I experiencing the same error.

diabl0w, 2021-05-02:
Same issue here... I've tried multiple BlueFlash versions, multiple csr-spi-ftdi versions, multiple hc-05 chips from different vendors (some were newer versions some were older versions) tried multiple different ftdi232r and ft232h modules. Ive spent a lot of money trying to get this to work and replaced every component multiple times and nothing is working

derskiy, 2022-11-29:
i have the same error, has anyone found a solution?