lorf/csr-spi-ftdi #57: Pure SPI?

BOOM-Mentoring, 2021-01-13:
How to make an SPI-only solution, that is, no USB -- A tool to program the CSR or QCC chips directly with SPI commands, say from an Arduino. 

Which parts of this code would be needed?

Is a command map available?

lorf, 2021-01-14:
Hi! 

Csr-spi-ftdi only implements transport layer of the protocol, You would need to implement it in Arduino (see https://github.com/Frans-Willem/CsrSpiDrivers). On top of that You would need to implement functionality you want (e.g. flashing firmware, controlling the device with BCCMD or configuring it with PS Keys).

Here are some pointers:

* Protocol analysis: https://web.archive.org/web/20190116215601/http://jernej87.blogspot.com/2012/10/csrs-usb-programmer-protocol-analysis.html
* Flash dumper: https://github.com/lorf/csrprogrammer
* BlueCore 4 flash programmer: https://web.archive.org/web/20180125112520/http://members.efn.org/~rick/work/rpi.csr.html
* BlueZ stack has BCCMD implementation over other transports: https://github.com/bluez/bluez/blob/utils-3.36/tools/bccmd.c
* Some useful information may be found by searching for "BlueSuiteSource_V2_5" and "BlueCore BCCMD Commands".

BOOM-Mentoring, 2021-01-14:
Thank you, Lorf!

Would you be willing to volunteer a little more time to help us solve this? That would be fantastic, and i'd be super grateful. You’d be helping us uplift teens in-crisis. We don’t have the expertise in-house to follow those awesome clues you shared.

The BOOM is a Resident at Autodesk, we won awards from Protolabs and Not Impossible Awards. We're a member of Engineers Without Borders. We've been featured in Make Magazine.

https://boo.ma/makezine

Looking forward to hearing from you.

Thx,
John

John Weiss, Director
The BOOM Mentoring
theboom.org<https://theboom.org>

Read about us: boo.ma/makezine<https://boo.ma/makezine>
Like us: facebook.com/bayviewboom<https://facebook.com/bayviewboom>
Follow us: twitter.com/bayviewboom<https://twitter.com/bayviewboom>
Donate: boo.ma/donate<https://boo.ma/donate>

415-68-WHEEL (415-689-4335)
c/o Renaissance Entrepreneurship Center
1325-B Evans Avenue
San Francisco, CA 94124

A project of Empowerment Works 501(c)3 Nonprofit Corporation
empowermentworks.org<https://empowermentworks.org>

BOOM-Mentoring, 2021-01-14:
i found some additional info that might be helpful, but i'm in over my head. 

https://linuxcommandlibrary.com/man/bccmd

https://himself.files.wordpress.com/2013/05/bccmd-commands.pdf

https://linux.die.net/man/8/bccmd

http://read.pudn.com/downloads158/ebook/704418/blab-ug-008PbPSTool-user-guide.pdf
see 2.1.2 Connecting PSTool Using Other Transport Interfaces 

https://bitbucket.org/primusinterpares/public/src/master/CSR_HostBoot/

https://cdn.sparkfun.com/datasheets/Wireless/Bluetooth/CSR-BC417-datasheet.pdf
see 11.7.1 Instruction Cycle

https://titanwolf.org/Network/Articles/Article?AID=d2eb856a-54c5-4d3b-afff-e50ca26e7d4c#gsc.tab=0

http://www1.futureelectronics.com/doc/UNIGEN/UGWC621RSMA133.pdf

doiiido, 2023-09-20:
Hi, I've found a bunch of projects that ported the CSR USB to SPI to other boards:
https://github.com/Frans-Willem/CsrUsbSpiDeviceRE
https://github.com/jkent/CsrUsbSpiDeviceRE
https://github.com/piskorzj/CsrUsbSpiDeviceRE-psoc5
I'll be working on it trying to port it to an ESP32 with the arduinoIDE.
Hope I bring good news soon!