lorf/csr-spi-ftdi #58: Faulty ST232RL?

felystar, 2021-02-20:
Hello guys,

I bought the FT232RL programmer (chinese version) to use it with an HC-05 chip. I followed the guide to install the BlueSuite software, the csr-spi-ftdi driver and plugged my FT232RL to the computer, and when trying to use BlueFlash I was getting the following error:

[image]

I searched online but I was only seeing a close error "invalid control data" instead of "getting control data". I found that it could be related to using USB 3.0 ports with counterfeit versions of the programmer, so I switched to a different computer with USB 2.0 ports, but the problem was the same. I even tried with a different USB cable. I kept looking for the solution and found that some people have devices with wrong vendor IDs, but mine has the right IDs:

[image]

I have tried different BlueSuite and csr-spi-ftdi driver versions, but the problem never changed. I found out about BlueFlashCmd and this is the error that I am getting:

```
> BlueFlashCmd.exe -chipver
blueflashcmd, version 2.5.0.93 Release
Copyright (C) 2002-2012, Cambridge Silicon Radio Ltd.

21:23:42.674586: all:spi.c:558:spi_init: csr-spi-ftdi 0.5.3, git rev 80b2ad0
21:23:42.957024: err:spi.c:166:spi_ftdi_xfer: ***************************************************
21:23:42.957024: err:spi.c:167:spi_ftdi_xfer: Lost 1 of 1 bytes of data in transit
21:23:42.959218: err:spi.c:168:spi_ftdi_xfer: Probably a counterfeit FT232RL in USB3.0 socket.
21:23:42.962034: err:spi.c:169:spi_ftdi_xfer: Try to plug programmer into USB 2.0 socket.
21:23:42.962874: err:spi.c:170:spi_ftdi_xfer: ***************************************************
21:23:43.013464: err:spi.c:166:spi_ftdi_xfer: ***************************************************
21:23:43.013464: err:spi.c:167:spi_ftdi_xfer: Lost 86 of 86 bytes of data in transit
21:23:43.016215: err:spi.c:168:spi_ftdi_xfer: Probably a counterfeit FT232RL in USB3.0 socket.
21:23:43.018564: err:spi.c:169:spi_ftdi_xfer: Try to plug programmer into USB 2.0 socket.
21:23:43.019570: err:spi.c:170:spi_ftdi_xfer: ***************************************************
21:23:43.021098: err:basics.cpp:481:spifns_sequence_read: Unable to start read (getting control data)
21:23:48.024989: err:spi.c:118:spi_ftdi_xfer: FTDI: write data failed: [-1] usb bulk write failed
21:23:48.025172: err:basics.cpp:481:spifns_sequence_read: Unable to start read (getting control data)
==============================================================================
ERROR: Error detecting chip type (Unable to start read (getting control data))
==============================================================================
21:23:53.092979: err:spi.c:118:spi_ftdi_xfer: FTDI: write data failed: [-1] usb bulk write failed
Failed
```

It keeps showing the message of USB 3.0 ports even though it is connected to USB 2.0 ports, and I have tried with 4 different computers with W10 and W7 (x64 bits).

I have never got to go beyond this error message. And I am starting to think about buying a new unit, but I wanted to check here just in case there is anything I've missed.

P.S.: I am testing the chip using "BlueFlashCmd.exe -chipver" using only the programmer with the USB cable, no HC-05 connections. I assume this is ok.

Thanks in advance for your help,
Felipe

Lalarian, 2021-06-06:
I can confirm this. Having teh exact same problem, with the same version of the chip (A50285BI)

552eden, 2024-12-31:
anyone manage to fix this?