lorf/csr-spi-ftdi #59: Modules on QCC300x chips

kiterservice, 2021-03-13:
This is not issue, just some info about chinese modules on QCC3003, QCC3005, and QQC3008 chips (RRD-305, RRD-303, BTM307, BTM308). 
1. Csr-spi-ftdi driver and ft232rl works on this modules only for PStool and Universal Parametr Manager. ADK configuration tool does not work. ft2232rl does not work at all. 
2. This modules has USB support, but only for charging and usb audio. Seems usb programming was disabled on chip firmware level (ahi sets to spi). 
3. For correct work with Csr-spi-ftdi driver pin EN should NOT be connected to VBAT. Pin PCM should be connected to pin  1.8V.

kiterservice, 2021-03-25:
Hello, it's me agane, guy with wierd english language. Senks to google translate.
Anabling usb on RRD-305 module for ADK Configuration tool. It's dirt hack, but if you have not csr-usb-spi adaptor, maybe it's 
helps. You doing it on your own risck
Metod consists in replasing original dump with SPI support on dump with USB support. 
You need instaled ADK_QCC300x.WIN.1.0.167 (if you are reading this probably you have).
-1. Download dump https://drive.google.com/drive/folders/1HUHScPHVU-FbnREi0fY9V9FPHyFZ0DCI?usp=sharing In folder you find two a banch of files with names starts from usb and spi. Copy files starts from usb to /ADK_QCC300x.WIN.1.0.167\tools\bin
0. Preparation as said in instruction on main page of this project
1. Run windows console (cmd) and change dirrectory to /ADK_QCC300x.WIN.1.0.167\tools\bin
2. Run nvscmd.exe dump original.ptn - it will show a lot of errors and as result after few minuts makes dump of you chip
3. Run nvscmd.exe burn usb.ptn all - it will show a lot of errors and as result write new dump on chip
4. Unsolder ft232 from you module. Solder wires from usb G to GND, D+ to USB-P D- to USB-N V+ to VBUS  and test connections.

LuthfiZXC, 2021-05-30:
> Hello, it's me agane, guy with wierd english language. Senks to google translate.
> Anabling usb on RRD-305 module for ADK Configuration tool. It's dirt hack, but if you have not csr-usb-spi adaptor, [...]

How to enable USB Programming on QCC3008 ? I tried ur dump file and it didn't work , the log said it has different partition size

kiterservice, 2021-05-30:
> How to enable USB Programming on QCC3008 ? I tried ur dump file and it didn't work , the log said it has different partition size

Unfortunately, in your case, i don't know  any way, exept by using csr-spi programmer. My dump work only with Chinese rrd-305 module based on qcc3005 chip. You module based on qcc3008 chip and has different flash rom with different partition table.
And, speeking honestly, usb connection almost useless. Pstool would not work with usb, for sound settings mutch easier using android app GAIA for qualcomm bluetooth chip.

RGBlu3-pro, 2021-05-31:
I have a little problem.
[image]

RGBlu3-pro, 2021-06-01:
On my second module your method works perfectly. I can do everything in ADK using only USB cable. Thank you so much :)

SkinnyDonut, 2023-01-20:
because i will mess up, i do the wiring with the default pins(from readme file) but i _dont connect pin en to vbat_ of the RRD-305 (basically dont short them?)
And _Pin PCM should be connected to pin 1.8V_ means that i have to short pcm and 1v8 on rrd-305?

Pic for help
[image]

djskomy, 2023-06-19:
Hi, can you re-upload the firmware files? link has expired.

riccardocagnasso, 2023-09-07:
> Hi, can you re-upload the firmware files? link has expired.

Bump I would really like the firmware too

djskomy, 2023-09-07:
I found it here: https://disk.yandex.ru/d/toQ3deXyBR-dMw
It's a part of some audio project: https://cxem.net/sound/amps/amp252.php

AntonioBanditoz, 2023-11-14:
Hello everyone I faced the problem of programming the RRD305 module, built a printed circuit board for this module, also purchased a CSR USB SPI module, but it cannot connect to the module in the ADK program, an error ERROR DETECTING CHIP TYPE (READ FAILED ON USB SPI) appears, the electrical wiring diagram of the module was provided to me by the seller. In the first post, you said that you don't need to connect the POWER_EN pin to VBAT, does this also apply to the original programmer?

SkinnyDonut, 2023-11-14:
Check the RRD305 documentation, if I remember correctly, one pin requires
5v+ (don't remember at all which pin, check the Pinout descriptions in the
documentation) in order to get it ready for programming

AntonioBanditoz, 2023-11-14:
> Проверьте документацию RRD305, если я правильно помню, для одного вывода требуется напряжение 5 В + (совсем не помню, для какого вывода, проверьте описания распиновок в документации), чтобы подготовить его к программированию
> […](#)

I checked from 3.3 to 4.2v, below is a link to the documentation, I think the link should open

https://szytf.yuque.com/sz0hxn/vzvo0a/gsfcwx?spm=a2g2w.orderlist.0.0.79ad4aa6dDNXnv

SkinnyDonut, 2023-11-14:
https://www.diyaudio.com/community/threads/csr8675-programming-guide-w-software-and-tons-of-csr-info.349336/page-8
Check comment #142

AntonioBanditoz, 2023-11-14:
Yes, I have the same connection scheme, the pin VS also tried to connect to 1.8v, the error is exactly the same. I am using ADK 1.0.167

SkinnyDonut, 2023-11-14:
Have you tested with older version?
I have formated my drive and don't remember the version I used
Check if the programmer has correct drivers

AntonioBanditoz, 2023-11-15:
Ok I'll try even older versions. Did you not understand about the drivers? They are also installed bundled with programs.

SkinnyDonut, 2023-11-15:
I remember that I had some problems with the drivers because I had used
zadig
Test with older versions and tell me

AntonioBanditoz, 2023-11-15:
you misunderstood me a little, I have an original csr usb spi programmer, I don’t need to replace the driver

AntonioBanditoz, 2023-11-19:
Thank you very much! installing the ADK QCC300X V1.0 version helped me.