lorf/csr-spi-ftdi #64: FTDI  recovery

omerguzelelectronicguy, 2022-01-08:
Hello I was trying to upgrade the firmware of HC05 but couldn't achieve. Now I want to make FTDI as its original version. Is it possible to do?

ShaSha146, 2022-01-13:
What changes did you make to your FTDI breakout board?

omerguzelelectronicguy, 2022-01-13:
I install the LibUSBK firmware to FTDI as stated in the [https://youtu.be/y8PcNbAA6AQ?t=341](url) video. Before I update the firmware, there is original firmware inside of it.

ShaSha146, 2022-01-13:
The LibUSBK is a driver not the firmware that runs on the FTDI breakout board, using the zadig tool should just install the driver to your computer. If you want to remove the driver you can uninstall the device and driver from your Device Manager if you're on Windows.

omerguzelelectronicguy, 2022-01-13:
So,I didn't change anything in FTDI. It wasn't mine, that's why I wanted to fix it before give back. I can give back it as it is. Thank you so much.

closed 2022-01-13 by omerguzelelectronicguy